open-power/op-build #2138: Build fails with `Could not find enumeration with ID of IBSCOM_MCS_BASE_ADDR`

paulmenzel, 2018-05-25:
```
$ git describe
v2.0-23-g5c16515
$ git log --oneline -1
5c16515 (HEAD -> master, origin/master, origin/HEAD) skiboot v6.0.1-27-g34e9c3c1edb3
$ . op-build-env
$ op-build firestone_defconfig
$ op-build
[…]
# creating the targeting binary
/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl --hb-xml-file=/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml --fapi-attributes-xml-file=/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/fapiattrs.xml --src-output-dir=/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/ --img-output-dir=/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/ --vmm-consts-file=/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/vmmconst.h --noshort-enums --bios-xml-file=/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"FIRESTONE_bios.xml" --bios-schema-file=/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/bios.xsd --bios-output-file=/home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/"firestone"_bios_metadata.xml
Could not find enumeration with ID of IBSCOM_MCS_BASE_ADDR
 at /home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 1276.
        main::getEnumerationType(HASH(0xe62494a3ff0), "IBSCOM_MCS_BASE_ADDR") called at /home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 4727
        main::packSingleSimpleTypeAttribute(SCALAR(0xe6248e0a018), REF(0xe6248e09fa0), REF(0xe6248e09fb8), "uint64_t", "0x000000000000000N") called at /home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 4807
        main::packAttribute(HASH(0xe62494a3ff0), HASH(0xe62495b5118), "0x000000000000000N") called at /home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 5510
        main::generateTargetingImage("/home/joey/src/op-build/output/host/powerpc64le-buildroot-"..., HASH(0xe62494a3ff0), HASH(0xe6248c18438), 0, SCALAR(0xe6248c37d10), SCALAR(0xe6248c37d28), SCALAR(0xe6248c37c98)) called at /home/joey/src/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 380
package/pkg-generic.mk:223: recipe for target '/home/joey/src/op-build/output/build/machine-xml-2494a430656504d014780000adbc2c2e7cfc54ab/.stamp_built' failed
make[1]: *** [/home/joey/src/op-build/output/build/machine-xml-2494a430656504d014780000adbc2c2e7cfc54ab/.stamp_built] Error 255
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2
make: Leaving directory '/home/joey/src/op-build/buildroot'
```

stewart-ibm, 2018-05-28:
Does this clear up if you do a clean build? i.e. `rm -rf output/*` before doing the `op-build firestone_defconfig && op-build` ?

usually this is because of a mismatch in the build directory between hostboot and other components such as the XML.

That whole build system is (unfortunately), incredibly fragile.

paulmenzel, 2018-05-28:
Well, I build from a clean checkout, so there shouldn’t have been a mismatch. I did `git clean -dfx` and tried again, and it’s still reproducible.

pridhiviraj, 2018-05-28:
May be similar to #871 .

paulmenzel, 2018-05-28:
On 05/28/18 13:58, Paidipeddi Pridhiviraj wrote:
> May be similar to #871 .

Looks like the same issue. Unfixed over a year. :(

I am closing this, and following #871.