open-power/op-build #2170: Support runtime init changes

stewart-ibm, 2018-06-05:
Things like the Spectre workarounds and some firmware features can possibly be changed at runtime by winkle/stop11-ing the core and bringing them back with new inits.

We currently don't have any method to do this at runtime, and require a re-IPL to do them. Perhaps we don't need the IPL for some of them.

dcrowell77, 2018-06-05:
This has been discussed, some high-level designs have even been proposed.  At a high-level this is no different from the Concurrent Core Init (CCI) that we already support during a concurrent code update in our enterprise products.  Unfortunately things get quite complicated when you start talking about a high number of variations.  The work to make this happen also introduces major incompatibilities between SBE/HCODE/HB/OPAL/PHYP that have to be managed.

stewart-ibm, 2018-06-06:
Dan <notifications@github.com> writes:
> This has been discussed, some high-level designs have even been
> proposed.  At a high-level this is no different from the Concurrent
> Core Init (CCI) that we already support during a concurrent code [...]

Yeah. It's not a simple thing currently and I'd probably say we should
clean up a bunch of those compatibility issues before we would attempt
to ship anything.

I was thinking of tracking it here as a possible future thing and we can
think about sizing of it and where it fits in priorities.

-- 
Stewart Smith
OPAL Architect, IBM.