open-power/op-build #2172: Remove need for hostboot-binaries

stewart-ibm, 2018-06-05:
Anything blob-like is a bug.

We still depend on some things in hostboot-binaries.

We shouldn't depend on them.

madscientist159, 2019-04-23:
Any progress on this?  HCODE in particular should be fully freed up by now.

dcrowell77, 2019-04-25:
The scanring inits (p9n.hw.rings.bin and p9n.hw.overlays.bin) are still binaries.  They cannot be generated inside op-build because they are based on confidential hardware specifications.

madscientist159, 2019-04-25:
@dcrowell77 My understanding was that we already got the go-ahead to release the source for those, that there were just tooling concerns at this point.  See https://git.raptorengineering.com/git/hcode-next/ for the source files as of DD2.2, just waiting on completion of the tooling.

Note that the tooling is something that we can work on if needed, we stopped because our understanding was that IBM would be releasing tooling.  We can pick this work back up if needed.

dcrowell77, 2019-04-26:
I am not familiar with that hcode-next work.  Maybe @jahunsbe can comment here.