open-power/op-build #2225: Pettitboot translations broken with Buildroot 2018.05-op-build

sammj, 2018-07-13:
After commit https://github.com/open-power/op-build/commit/35ac8f6ec350dd8f3cbb4348c7663c3140140015 we appear to be missing some locale support in Petitboot:

```
 Petitboot Language Selection
 ──────────────────────────────────────────────────────────────────────────────
  Language       ( ) Deutsch
                 (*) English
                 ( ) Unable to display text in this locale (C)
                 ( ) Unable to display text in this locale (C)
                 ( ) Italiano
                 ( ) Unable to display text in this locale (C)
                 ( ) Unable to display text in this locale (C)
                 ( ) Unable to display text in this locale (C)
                 ( ) Unable to display text in this locale (C)
                 ( ) Unable to display text in this locale (C)
                 ( ) Unable to display text in this locale (C)
                 [    OK    ]  [  Cancel  ]
```

Investigating..

sammj, 2018-07-18:
This is caused by the update from glibc-2.26 to glib-2.27. From the release notes:
```
Deprecated and removed features, and other changes affecting compatibility:

* Statically compiled applications attempting to load locales compiled for the
  GNU C Library version 2.27 will fail and fall back to the builtin C/POSIX
  locale.  The reason for this is that the addition of the new "%OB" and "%Ob",
  support for two grammatical forms of the month names, also extends the locale
  data binary format.  Static applications needing locale support must be
  recompiled to match the runtime and data they are deployed with.
```
The same issue is discussed in this Buildroot bug: https://bugs.busybox.net/show_bug.cgi?id=11096
As the bug says downgrading glibc will temporarily fix it, but I'll see if there's a way to force Buildroot's `localedef` to be compiled against 2.27

sammj, 2018-07-24:
I've pushed a proposed fix for this here: https://github.com/sammj/buildroot/commit/b9ba0ff5c1f6d942f24a7a8a30946262e1a640a6
This does a full glibc compile for the host which takes a little longer but avoids having to patch the glibc build to only build localedef.

shlimeng, 2018-07-24:
applied the patch from sammj/buildroot@b9ba0ff, but build failed on openpower system with ubuntu distro.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial
$ gcc --version
gcc (Ubuntu/IBM 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
configure: error: ***  binary128 floating point type (GCC >= 6.2) is required on powerpc64le.
package/pkg-generic.mk:213: recipe for target '/home/op-build/output/build/host-localedef-glibc-2.27-57-g6c99e37f6fb640a50a3113b2dbee5d5389843c1e/.stamp_configured' failed
make[1]: *** [/home/op-build/output/build/host-localedef-glibc-2.27-57-g6c99e37f6fb640a50a3113b2dbee5d5389843c1e/.stamp_configured] Error 1
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2
make: Leaving directory '/home/op-build/buildroot'
```

shlimeng, 2018-07-24:
This issue can be fixed with the patch from sammj/buildroot@b9ba0ff and update host gcc to `6.4.0` (GCC >= 6.2) when build pnor.
result shows below:
```
 Petitboot Language Selection
 ──────────────────────────────────────────────────────────────────────────────

  Language       (*) Deutsch
                 ( ) English
                 ( ) Español
                 ( ) Français
                 ( ) Italiano
                 ( ) 日本語
                 ( ) 한국어
                 ( ) Português/Brasil
                 ( ) Русский
                 ( ) 简体中文
                 ( ) 繁體中文

                 [    OK    ]  [  Cancel  ]
```

sammj, 2019-02-18:
There's a second patch on the Buildroot list that runs with this idea but uses a glibc patch like ptxdist does to only build `localedef` which saves some build time. Ideally if that goes upstream it will fix this issue.
https://patchwork.ozlabs.org/patch/1008984/

sammj, 2019-03-14:
This is a combination of my localdef patch and a following patch by Paul to avoid building the whole of glibc: https://github.com/sammj/buildroot/commit/bdfd9ef522373d41767308fffaf497ae44dbabb5
While we try and upstream something, is it worth carrying this in our branch so that we have translations? @shenki ?

shenki, 2019-03-14:
Yes we can carry that. We're blocked on @stewart-ibm merging the latest buildroot bump, which is in turned blocked on some CI infrastructure updates.

I have appled the patch to our 2019.02-op-build tree now. It will come "for free" once we merge https://github.com/open-power/op-build/pull/2666.

sammj, 2019-06-19:
I've updated the above patch to match glibc 2.29 and posted it to the mailing list: https://github.com/sammj/buildroot/commit/b02039d9edcaf4adca38c96a9eb2f65122ff13fb