open-power/op-build #2230: Could not find enumeration with ID of IBSCOM_MCS_BASE_ADDR

shenki, 2018-07-18:
On the Ubuntu 18.04 container that is used for op-build CI on openpower.xyz, we see this failure when building for Palmetto:

```
/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl --hb-xml-file=/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml --fapi-attributes-xml-file=/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/fapiattrs.xml --src-output-dir=/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/ --img-output-dir=/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/ --vmm-consts-file=/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/vmmconst.h --noshort-enums --bios-xml-file=/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"PALMETTO_bios.xml" --bios-schema-file=/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/bios.xsd --bios-output-file=/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/"palmetto"_bios_metadata.xml
Could not find enumeration with ID of IBSCOM_MCS_BASE_ADDR
 at /var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 1276.
	main::getEnumerationType(HASH(0x561ca21a94d8), "IBSCOM_MCS_BASE_ADDR") called at /var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 4727
	main::packSingleSimpleTypeAttribute(SCALAR(0x561ca1b08ce0), REF(0x561ca1b08c68), REF(0x561ca1b08c80), "uint64_t", "0x000000000000000N") called at /var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 4807
	main::packAttribute(HASH(0x561ca21a94d8), HASH(0x561ca22b8610), "0x000000000000000N") called at /var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 5510
	main::generateTargetingImage("/var/lib/jenkins/workspace/openpower-op-build/label/slave/tar"..., HASH(0x561ca21a94d8), HASH(0x561ca13f3318), 0, SCALAR(0x561ca18fa6a0), SCALAR(0x561ca18fa6b8), SCALAR(0x561ca18fa628)) called at /var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl line 380
package/pkg-generic.mk:223: recipe for target '/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/build/machine-xml-c6f563966e9fadc4fb60194c064b2310c9b916b1/.stamp_built' failed
make: *** [/var/lib/jenkins/workspace/openpower-op-build/label/slave/target/palmetto/op-build/output/build/machine-xml-c6f563966e9fadc4fb60194c064b2310c9b916b1/.stamp_built] Error 255
```

dcrowell77, 2018-07-18:
What changed here?  Nothing in our code has changed for Palmetto in a long time.  Is this a new perl version maybe?

shenki, 2018-07-18:
Yes, I suspect a newer perl version or perl library version as exposed an issue in the existing scripts.

shenki, 2018-07-18:
I have a proposed fix:

 https://github.com/open-power/palmetto-xml/pull/19

closed 2018-07-20 by stewart-ibm