open-power/op-build #2415: Useful skiboot artefacts are not copied to output directory

shenki, 2018-10-25:
When building we end up with `skiboot.lid` and `skiboot.lid.bin`.

Ideally we would end up with `skiboot.elf`, and `skiboot.lid.xz` for p8, or `skiboot.lid.xz.stb` for p9.

dcrowell77, 2018-10-25:
I see skiboot.elf and skiboot.map included as part of output/images/host_fw_debug.tar.

stewart-ibm, 2018-12-06:
It gets weird here.

So, we assemble a `skiboot.lid.xz` as part of building skiboot - we build with `xz -9` and CRC32. It turns out that `update_images.pl` in the pnor/ repo constructs its *own* compressed skiboot with the default xz compression level (6). So the skiboot build artifact is **not** being used.

Additionally, even though we produce the secure boot header with the imprint keys, the genPnorImages.pl script in the **hostboot** repository is what generates the secureboot header for the skiboot.lid that ends up in the PAYLOAD partition (via scripts in the pnor repo).

stewart-ibm, 2018-12-06:
I'll assign this to myself as I'm reworking the trainwreck that is PNOR assembly currently and this issue should be resolved as a side-effect of it.