open-power/op-build #2519: P8 mambo does not run on new distros

shenki, 2018-12-12:
Debian unstable has removed OpenSSL 1.0.0 from the archive, as it is now 9 years old and has been replaced by newer releases.

The published P8 mambo/systemsim 1.0-2 release ( links against openssl, and therefore does not run:
```
/opt/ibm/systemsim-p8/bin/systemsim-pegasus: error while loading shared
libraries: libcrypto.so.1.0.0: cannot open shared object file: No such
file or directory
```
It would be great if we could have a newer binary built that runs on newer distros

shenki, 2018-12-12:
I have reached out to Brad Thomasson at IBM who handles the public mambo releases to see if we can get a new binary published.

stewartsmith, 2020-11-08:
Since POWER8 and palmetto support was removed in 40f804058414ddf4518ac07866bb58af74db20a2, this can probably be closed now.

closed 2020-11-09 by dcrowell77