open-power/op-build #2660: 4.20 skiroot kernel crashes

shenki, 2019-03-04:
> ```
> ./op-test --config-file bstnbmc008.conf --host-pnor ~/p9dsu.pnor 
>  --run testcases.OpTestInbandIPMI.SkirootBasicInbandIPMI
> ```
> After it passed connected to SoL and found it in xmon.

Kernel said:
```
[   27.346994] Unable to handle kernel paging request for data at address 0x1ffdbc0004
[   27.346997] Faulting instruction address: 0xc00000000047d72c
```
And the console had:
```
ipmitool event 3   
Sending SAMPLE event: Memory - Correctable ECC                                                      
cpu 0x2c: Vector: 300 (Data Access) at [c000001ff74d36d0]                                           
    pc: c00000000047d72c: __do_softirq+0xd4/0x2b0                                                   
    lr: c00000000047d8ac: __do_softirq+0x254/0x2b0                                                  
    sp: c000001ff74d3960                                                                            
   msr: 9000000000009033                                                                            
   dar: 1ffdbc0004 
 dsisr: 42000000   
  current = 0xc000001ff73cca80                                                                         
  paca    = 0xc000001ffffde600     irqmask: 0x03        irq_happened: 0x01                          
    pid   = 0, comm = swapper/44
Linux version 4.20.6-openpower1 (jenkins@bstn004p1) (gcc version 6.4.0 (Buildroot 2018.11.2-00011-g7
ded521)) #2 SMP Mon Feb 11 05:22:17 UTC 2019
enter ? for help
[c000001ff74d3a50] c00000000007c630 irq_exit+0x7c/0xd4
[c000001ff74d3a70] c00000000002bdd4 doorbell_exception+0x88/0xa4
[c000001ff74d3aa0] c000000000009b74 h_doorbell_common+0x124/0x130
--- Exception: e81 (Hypervisor Doorbell) at c000
00000000a8fc replay_interrupt_return+0x0/0x4
[link register   ] c000000000012a80 arch_local_irq_restore+0x5c/0x80
[c000001ff74d3da0] 0000000000462eff (unreliable)
[c000001ff74d3dc0] c000000000334334 cpuidle_enter_state+0x1a0/0x20c
[c000001ff74d3e10] c0000000000a0ee0 call_cpuidle+0x6c/0x74
[c000001ff74d3e30] c0000000000a12f0 do_idle+0x250/0x280
[c000001ff74d3ec0] c0000000000a14ac cpu_startup_entry+0x30/0x34
[c000001ff74d3ef0] c00000000002e988 start_secondary+0x3bc/0x528
[c000001ff74d3f90] c00000000
000ac70 start_secondary_prolog+0x10/0x14
2c:mon>
```

@mpe has some analysis of the crash in https://github.com/open-power/op-build/pull/2598#issuecomment-462275960 .

mpe, 2019-03-07:
Could this be related to https://github.com/open-power/supermicro-openpower/issues/497 ?

Over there turning on slub_debug seems to have helped get closer to a root cause.

shenki, 2019-04-30:
We moved to 5.0 and have not seen this crash recently. Closing for now.

closed 2019-04-30 by shenki