open-power/op-build #2760: op-build master builds failing on RHel7

fspbld, 2019-04-05:
...
> checking if /bin/gcc is sufficient to build libc... yes
> checking for x86_64-pc-linux-gnu-nm... /bin/nm
> checking for python3... no
> checking for python... python
> configure: error: 
> *** These critical programs are missing or too old: as GNU ld
> *** Check the INSTALL file for required versions.
> make[1]: ***

This error seems to have started with the buildroot update in commit, 
e4b1465 Merge pull request #2732 from shenki/buildroot-bump

fspbld, 2019-04-05:
This fails for zaius, romulus, witherspoon (not surprisingly, since it doesn't even get far enough to really do anything platform-specific).  :}   The package it seems to fail in while processing is host-localedef-glibc-2.28.

sammj, 2019-04-07:
Please post the output of
`cat /etc/os-release`
and
`ld --version`.

We just went through some pain getting the CI build machines up to date to satsify some build dependencies and AFAIK they're also running RHEL 7. Why there's a difference here I'm not sure yet, maybe different minor versions of RHEL 7?

fspbld, 2019-04-08:
cat /etc/os-release
NAME="Red Hat Enterprise Linux Workstation"
VERSION="7.2 (Maipo)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="7.2"
PRETTY_NAME="Red Hat Enterprise Linux Workstation 7.2 (Maipo)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:redhat:enterprise_linux:7.2:GA:workstation"
HOME_URL="https://www.redhat.com/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 7"
REDHAT_BUGZILLA_PRODUCT_VERSION=7.2
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="7.2"

ld --version
GNU ld version 2.23.52.0.1-55.el7 20130226
Copyright 2013 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

fspbld, 2019-04-08:
I did ask our admin guy about what he might have done for the CI machines and nothing seemed to have clicked with him (or maybe someone was just debugging the issues on the machines and telling him, exactly, what needed to be updated and he didn't know why the request was coming in?)

fspbld, 2019-04-08:
Looks like CI machines bumped up to RH7.4 - is that the new required minimum level?

fspbld, 2019-04-08:
we are taking a machine up to RH7.4 ... will try that ..and keep you posted

stewart-ibm, 2019-04-09:
Pam Eggler <notifications@github.com> writes:
> cat /etc/os-release
> NAME="Red Hat Enterprise Linux Workstation"
> VERSION="7.2 (Maipo)"
> ID="rhel"
> ID_LIKE="fedora"
> VERSION_ID="7.2"
> PRETTY_NAME="Red Hat Enterprise Linux Workstation 7.2 (Maipo)" [...]

Here's the configure snippet:

# Accept binutils 2.25 or newer.                                                                                                  
AC_CHECK_PROG_VER(AS, $AS, --version,
                  [GNU assembler.* \([0-9]*\.[0-9.]*\)],
                  [2.1[0-9][0-9]*|2.2[5-9]*|2.[3-9][0-9]*|[3-9].*|[1-9][0-9]*],
                  AS=: critic_missing="$critic_missing as")

if test -n "`$LD --version | sed -n 's/^GNU \(gold\).*$/\1/p'`"; then
  # Accept gold 1.14 or higher                                                                                                    
  AC_CHECK_PROG_VER(LD, $LD, --version,
                    [GNU gold.* \([0-9][0-9]*\.[0-9.]*\)],
                    [1.1[4-9]*|1.[2-9][0-9]*|1.1[0-9][0-9]*|[2-9].*|[1-9][0-9]*],
                    LD=: critic_missing="$critic_missing GNU gold")
else
  AC_CHECK_PROG_VER(LD, $LD, --version,
                    [GNU ld.* \([0-9][0-9]*\.[0-9.]*\)],
                    [2.1[0-9][0-9]*|2.2[5-9]*|2.[3-9][0-9]*|[3-9].*|[1-9][0-9]*],
                    LD=: critic_missing="$critic_missing GNU ld")
fi

and we should be hitting that last one, which seems to think that we
want 2.25 or newer for 2.2x versions.

If I grab the latest centos:7 docker image, I get:

[root@636f5ab1c5a9 /]# ld --version
GNU ld version 2.27-34.base.el7
Copyright (C) 2016 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

Also, if I go for the 7.4 one listed on https://hub.docker.com/_/centos

[root@b250b1005f34 /]# ld --version
GNU ld version 2.25.1-31.base.el7 
Copyright (C) 2014 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

So yeah, the upgrade to RHEL7.4 should do the trick.

-- 
Stewart Smith
OPAL Architect, IBM.

fspbld, 2019-04-09:
yep, latest build for romulus worked

closed 2019-04-09 by fspbld