open-power/op-build #2764: TM doesn't work for DD2.3 due to missing cpu features

mikey, 2019-04-08:
With op-build v2.3 (via OP930), I get this with a modern linux kernel.
```
 %  qemu-system-ppc64  -nographic -M pseries  -enable-kvm 
qemu-system-ppc64: KVM implementation does not support Transactional Memory, try cap-htm=off
```
This error means Transactional Memory is not available to guests, which is wrong for DD2.3.

This is due to skiboot not advertising the required Transactional Memory workarounds for DD2.3.

This patch needs to be included to fix this:
  https://patchwork.ozlabs.org/patch/1075443/

mikey, 2019-04-12:
This patch is now in skiboot here:
https://github.com/open-power/skiboot/commit/24268c7662068c276396220a68bb32d17715ebba

sammj, 2019-05-31:
op-build-2.3.y is on v6.3.1, are we all good here?

mikey, 2019-05-31:
Yeah, we should be. I'll reopen if I hit it again.

closed 2019-05-31 by mikey