open-power/op-build #2816: Petitboot starts but guest kernels hang

madscientist159, 2019-04-23:
After updating the PNOR firmware stack to latest master, we're seeing a problem where skiroot loads and appears to function normally, but guest kernels hang early in the boot process.

Skiroot may not be 100% functional either, the following messages are shown before petiboot loads:

```
zImage starting: loaded at 0x0000000020011000 (sp: 0x00000000214bdee8)
Allocating 0x24e0520 bytes for kernel...
Decompressing (0x0000000000000000 <- 0x0000000020020000:0x00000000214bb600)...
Done! Decompressed 0x22f80d8 bytes

Linux/PowerPC load:
Finalizing device tree... flat tree at 0x214bf0a0
[   92.210952828,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xd0010080
[   92.210969191,6] IPMI: dropping non severe PEL event
[   92.211049606,7] UART: IRQ functional !
[    7.416415] IMC PMU (null) Register failed
```

Attempted boot with known good 4.18 kernel.  The logs consistently stop at the same location (system hangs):

```
SIGTERM received, booting...
[  177.116744] kexec_core: Starting new kernel
[    0.000000] dt-cpu-ftrs: setup for ISA 3000
[    0.000000] dt-cpu-ftrs: not enabling: system-call-vectored (disabled or unsupported by kernel)
[    0.000000] dt-cpu-ftrs: final cpu/mmu features = 0x0000f86f8f5fb1a7 0x3c006041
[    0.000000] radix-mmu: Page sizes from device-tree:
[    0.000000] radix-mmu: Page size shift = 12 AP=0x0
[    0.000000] radix-mmu: Page size shift = 16 AP=0x5
[    0.000000] radix-mmu: Page size shift = 21 AP=0x1
[    0.000000] radix-mmu: Page size shift = 30 AP=0x2
[    0.000000] radix-mmu: Initializing Radix MMU
[    0.000000] radix-mmu: Partition table (____ptrval____)
[    0.000000] radix-mmu: Mapped 0x0000000000000000-0x0000000001200000 with 64.0 KiB pages
[    0.000000] radix-mmu: Mapped 0x0000000001200000-0x0000000040000000 with 2.00 MiB pages
[    0.000000] radix-mmu: Mapped 0x0000000040000000-0x0000000800000000 with 1.00 GiB pages
[    0.000000] radix-mmu: Mapped 0x0000200000000000-0x0000200800000000 with 1.00 GiB pages
[    0.000000] radix-mmu: Process table (____ptrval____) and radix root for kernel: (____ptrval____)
[    0.000000] Linux version 4.18.0+ (root@build) (gcc version 8.2.0 (Debian 8.2.0-3)) #2 SMP Sun Aug 19 04:46:53 UTC 2018
[    0.000000] Found initrd at 0xc0000000039e0000:0xc0000000049f52d4
[    0.000000] OPAL: Found memory mapped LPC bus on chip 0
[    0.000000] Using PowerNV machine description
[    0.000000] bootconsole [udbg0] enabled
[    0.000000] CPU maps initialized for 4 threads per core
[    0.000000] -----------------------------------------------------
[    0.000000] ppc64_pft_size    = 0x0
[    0.000000] phys_mem_size     = 0x1000000000
[    0.000000] dcache_bsize      = 0x80
[    0.000000] icache_bsize      = 0x80
[    0.000000] cpu_features      = 0x0000f86f8f5fb1a7
[    0.000000]   possible        = 0x0000ffffcf5fb1a7
[    0.000000]   always          = 0x00000003800081a1
[    0.000000] cpu_user_features = 0xdc0065c2 0xaee00000
[    0.000000] mmu_features      = 0x3c006041
[    0.000000] firmware_features = 0x0000000110000000
[    0.000000] -----------------------------------------------------
[    0.000000] cma: Reserved 3280 MiB at 0x000020072f000000
[    0.000000] numa:   NODE_DATA [mem 0x7ff822300-0x7ff82bfff]
[    0.000000] numa:   NODE_DATA [mem 0x2007fca3e300-0x2007fca47fff]
[    0.000000] rfi-flush: mttrig type flush available
[    0.000000] barrier-nospec: using ORI speculation barrier
[    0.000000] stf-barrier: eieio barrier available
[    0.000000] Initializing IODA2 PHB (/pciex@600c3c0000000)
[    0.000000] PCI host bridge /pciex@600c3c0000000 (primary) ranges:
[    0.000000]  MEM 0x000600c000000000..0x000600c07ffeffff -> 0x0000000080000000
[    0.000000]  MEM 0x0006000000000000..0x0006003fffffffff -> 0x0006000000000000 (M64 #1..31)
[    0.000000]  Using M64 #31 as default window
[    0.000000]   512 (511) PE's M32: 0x80000000 [segment=0x400000]
[    0.000000]                  M64: 0x4000000000 [segment=0x20000000]
[    0.000000]   Allocated bitmap for 4088 MSIs (base IRQ 0xfe000)
[... 432 lines cut ...]
[    2.569124] pcieport 0031:02:06.0: enabling device (0545 -> 0547)
[    2.569542] pcieport 0031:02:07.0: enabling device (0545 -> 0547)
[    2.570168] dpc 0031:02:01.0:pcie208: DPC error containment capabilities: Int Msg #0, RPExt- PoisonedTLP+ SwTrigger+ RP PIO Log 0, DL_ActiveErr+
[    2.570358] dpc 0031:02:04.0:pcie208: DPC error containment capabilities: Int Msg #0, RPExt- PoisonedTLP+ SwTrigger+ RP PIO Log 0, DL_ActiveErr+
[    2.570617] dpc 0031:02:05.0:pcie208: DPC error containment capabilities: Int Msg #0, RPExt- PoisonedTLP+ SwTrigger+ RP PIO Log 0, DL_ActiveErr+
[    2.570894] dpc 0031:02:06.0:pcie208: DPC error containment capabilities: Int Msg #0, RPExt- PoisonedTLP+ SwTrigger+ RP PIO Log 0, DL_ActiveErr+
[    2.571084] dpc 0031:02:07.0:pcie208: DPC error containment capabilities: Int Msg #0, RPExt- PoisonedTLP+ SwTrigger+ RP PIO Log 0, DL_ActiveErr+
[    2.571284] pciehp 0031:02:04.0:pcie204: Slot #4 AttnBtn+ PwrCtrl+ MRL- AttnInd+ PwrInd+ HotPlug+ Surprise+ Interlock- NoCompl- LLActRep-
[    2.571486] pciehp 0031:02:05.0:pcie204: Slot #37 AttnBtn+ PwrCtrl+ MRL- AttnInd+ PwrInd+ HotPlug+ Surprise+ Interlock- NoCompl- LLActRep-
[    2.571580] pciehp 0031:02:04.0:pcie204: Slot(4): Card present
[    2.571740] pciehp 0031:02:06.0:pcie204: Slot #70 AttnBtn+ PwrCtrl+ MRL- AttnInd+ PwrInd+ HotPlug+ Surprise+ Interlock- NoCompl- LLActRep-
[    2.572068] pciehp 0031:02:07.0:pcie204: Slot #103 AttnBtn+ PwrCtrl+ MRL- AttnInd+ PwrInd+ HotPlug+ Surprise+ Interlock- NoCompl- LLActRep-
[    2.572476] hvc0: raw protocol on /ibm,opal/consoles/serial@0 (boot console)
[    2.572550] hvc0: No interrupts property, using OPAL event
[    2.573139] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    2.573646] Linux agpgart interface v0.103
[    2.575381] mousedev: PS/2 mouse device common for all mice
[    2.611656] rtc-opal opal-rtc: rtc core: registered rtc-opal as rtc0
[    2.614633] powernv-cpufreq: cpufreq pstate min 0x63 nominal 0x63 max 0x0
[    2.614696] powernv-cpufreq: Workload Optimized Frequency is enabled in the platform
[    2.637363] ledtrig-cpu: registered to indicate activity on CPUs
[    2.637878] NET: Registered protocol family 10
```

@stewart-ibm Is this something OzLabs has seen before?

madscientist159, 2019-04-24:
Also seeing this on a Linux 5.0 test kernel.  I note in both cases the hang occurs after two potentially relevant messages:

```
[    3.276232] hvc0: raw protocol on /ibm,opal/consoles/serial@0 (boot console)
[    3.276268] hvc0: No interrupts property, using OPAL event
```

```
[    3.314037] powernv-cpufreq: cpufreq pstate min 0x63 nominal 0x63 max 0x0
[    3.314068] powernv-cpufreq: Workload Optimized Frequency is enabled in the platform
```

sammj, 2019-04-26:
> After updating the PNOR firmware stack to latest master, we're seeing a problem where skiroot loads and appears to function normally, but guest kernels hang early in the boot process.

Could you describe what the original PNOR versions were prior to updating?

madscientist159, 2019-04-26:
> > After updating the PNOR firmware stack to latest master, we're seeing a problem where skiroot loads and appears to function normally, but guest kernels hang early in the boot process.
> 
> Could you describe what the original PNOR versions were prior to updating?

Original PNOR was:
* https://github.com/open-power/hostboot/commit/496156ffbd83372deb1f7a6e160d4fec3c525b5c
* https://github.com/open-power/skiboot/commit/67bdd1668e6a033eb5520105946ebb651f3adff2
* https://github.com/open-power/hcode/commit/699005f149f1f1eec45b20c06dc776684e4d6b0f
* https://github.com/open-power/sbe/commit/3c633a6c2cd626dafef63f40bfd611016b05ecc7

The fewer CPUs I bring online (via the guest kernel "maxcpus" parameter) the further the boot process gets -- 1 CPU (maxcpus=1) gets partway into systemd startup.

madscientist159, 2019-04-26:
Looks like "powersave=off" passed to the guest kernel works around the problem, the system boots and runs normally.  I know there was stop state trouble in the past leading to sporadic hangs after a few days of uptime, is there any chance it's back and leading to a more rapid hang?

a-barrera, 2019-04-26:
@madscientist159 that helps narrow things down. After looking at some of our internal testing, I think this is a dup of: https://bugzilla.linux.ibm.com/show_bug.cgi?id=177011

The problem appears to have been introduced by some recent CME changes, see below, that introduced the bug on systems running a Little Endian OS. The power management firmware team is working on a fix.

https://github.com/open-power/hcode/commit/f3d82eaa4d8c8181bb2f13b47f6ae67c1e553c1d

madscientist159, 2019-04-26:
@a-barrera Unfortunately that Bugzilla is inaccessible outside IBM...can you forward the pertinent information to my Email address?

Thanks!

madscientist159, 2019-04-26:
@a-barrera I can confirm reverting https://github.com/open-power/hcode/commit/f3d82eaa4d8c8181bb2f13b47f6ae67c1e553c1d works around the problem.

a-barrera, 2019-05-03:
@madscientist159 fix should be upstream now:

https://github.com/open-power/hcode/commit/cc0b781abd67673240c208df7c77d2d37906057f

yesyjim, 2020-04-08:
I am seeing this same issue on a power8 8335-GTB system. Looking at the fix it was implemented on power8, is there a workaround for this issue for p8?

stewartsmith, 2020-04-13:
> @mzipse I am seeing this same issue on a power8 8335-GTB system. Looking at the fix it was implemented on power8, is there a workaround for this issue for p8?

Does your problem resolve with "powersave=off" passed to the kernel?

yesyjim, 2020-04-13:
@stewartsmith  Where do I write the powersave=off ?  In the BMC?   

Here is the console log: 

--== Welcome to Hostboot hostboot-p8-c893515-pd6f049d/hbicore.bin ==--

  9.57412|ISTEP  6. 3 - host_init_fsi
  9.93291|ISTEP  6. 4 - host_set_ipl_parms
  9.93371|ISTEP  6. 5 - host_discover_targets
 76.11553|HWAS|PRESENT> DIMM[03]=AAAAAAAAAAAAAAAA
 76.11553|HWAS|PRESENT> Membuf[04]=CCCC000000000000
 76.11554|HWAS|PRESENT> Proc[05]=C000000000000000
 76.24722|ISTEP  6. 6 - host_gard
 78.70020|ISTEP  6. 7 - host_cancontinue_clear
 81.32563|ISTEP  6. 8 - proc_revert_sbe_mcs_setup
 81.41559|ISTEP  6. 9 - host_slave_sbe_config
 85.55716|ISTEP  6.10 - host_sbe_start
 85.57698|ISTEP  6.11 - proc_check_slave_sbe_seeprom_complete
 87.31947|ISTEP  6.12 - proc_cen_ref_clk_enable
 87.32149|ISTEP  6.13 - proc_xmit_sbe
 87.32204|ISTEP  7. 1 - proc_attr_update
 87.38034|ISTEP  7. 2 - proc_a_x_pci_dmi_pll_initf
 87.41926|ISTEP  7. 3 - proc_a_x_pci_dmi_pll_setup
 87.45850|ISTEP  7. 4 - proc_startclock_chiplets
 87.48483|ISTEP  7. 5 - proc_chiplet_scominit
 87.72998|ISTEP  7. 6 - proc_xbus_scominit
 87.90619|ISTEP  7. 7 - proc_abus_scominit
 87.92076|ISTEP  7. 8 - proc_pcie_scominit
 88.09694|ISTEP  7. 9 - proc_scomoverride_chiplets
 88.09789|ISTEP  8. 1 - fabric_erepair
 88.22173|ISTEP  8. 2 - fabric_io_dccal
 88.22859|ISTEP  8. 3 - fabric_pre_trainadv
 88.24662|ISTEP  8. 4 - fabric_io_run_training
 88.33513|ISTEP  8. 5 - fabric_post_trainadv
 88.33623|ISTEP  8. 6 - host_startprd_pbus
 88.87866|ISTEP  8. 7 - host_attnlisten_proc
 88.87948|ISTEP  8. 8 - proc_fab_iovalid
 88.94299|ISTEP  9. 1 - proc_build_smp
 89.29096|ISTEP  9. 2 - host_slave_sbe_update
 89.87060|ISTEP 10. 1 - host_prd_hwreconfig
 90.12607|ISTEP 10. 2 - cen_sbe_tp_chiplet_init1
 91.44204|ISTEP 10. 3 - cen_sbe_pll_initf
 91.44322|ISTEP 10. 4 - cen_sbe_pll_setup
 91.44423|ISTEP 10. 5 - cen_sbe_tp_chiplet_init2
 91.44523|ISTEP 10. 6 - cen_sbe_tp_arrayinit
 91.44871|ISTEP 10. 7 - cen_sbe_tp_chiplet_init3
 91.44988|ISTEP 10. 8 - cen_sbe_chiplet_init
 91.45105|ISTEP 10. 9 - cen_sbe_arrayinit
 91.45211|ISTEP 10.10 - cen_sbe_dts_init
 91.45527|ISTEP 10.11 - cen_sbe_initf
 91.45625|ISTEP 10.12 - cen_sbe_do_manual_inits
 91.46126|ISTEP 10.13 - cen_sbe_nest_startclocks
 91.46239|ISTEP 10.14 - cen_sbe_scominits
 91.49025|ISTEP 11. 1 - mss_getecid
 91.75737|ISTEP 11. 2 - dmi_attr_update
 91.75825|ISTEP 11. 3 - proc_dmi_scominit
 91.88551|ISTEP 11. 4 - dmi_scominit
 92.09269|ISTEP 11. 5 - dmi_erepair
 92.27562|ISTEP 11. 6 - dmi_io_dccal
 94.37575|ISTEP 11. 7 - dmi_pre_trainadv
 94.37714|ISTEP 11. 8 - dmi_io_run_training
 94.96918|ISTEP 11. 9 - dmi_post_trainadv
 94.97053|ISTEP 11.10 - proc_cen_framelock
 95.04036|ISTEP 11.11 - host_startprd_dmi
 95.04149|ISTEP 11.12 - host_attnlisten_cen
 95.04282|ISTEP 11.13 - cen_set_inband_addr
 95.04401|ISTEP 12. 1 - host_collect_dimm_spd
 95.13252|ISTEP 12. 2 - mss_volt
 95.21908|ISTEP 12. 3 - mss_freq
 95.24552|ISTEP 12. 4 - mss_eff_config
[... 245 lines cut ...]
[c000003fffe8bbc0] d00000000e3b4054 mlx5_cmd_comp_handler+0x10c/0x3c0 [mlx5_core]
[c000003fffe8bcb0] d00000000e3b4330 cmd_comp_notifier+0x28/0x44 [mlx5_core]
[c000003fffe8bcd0] c0000000000979b8 notifier_call_chain+0x7c/0xa8
[c000003fffe8bd20] d00000000e3b6abc mlx5_eq_async_int+0xd0/0x180 [mlx5_core]
[c000003fffe8bdc0] c0000000000bb7e4 __handle_irq_event_percpu+0x68/0x138
[c000003fffe8be70] c0000000000bb8e4 handle_irq_event_percpu+0x30/0x74
[c000003fffe8beb0] c0000000000bb990 handle_irq_event+0x68/0x94
[c000003fffe8bee0] c0000000000c0274 handle_fasteoi_irq+0xd8/0x18c
[c000003fffe8bf10] c0000000000ba58c generic_handle_irq+0x34/0x54
[c000003fffe8bf30] c00000000001332c __do_irq+0xc8/0xd0
[c000003fffe8bf90] c00000000001f144 call_do_irq+0x14/0x24
[c000000001297a30] c0000000000133cc do_IRQ+0x98/0xc4
[c000000001297a80] c000000000008c24 hardware_interrupt_common+0x124/0x130
--- Exception: 501 (Hardware Interrupt) at c00000000000a90c replay_interrupt_return+0x0/0x4
[link register   ] c000000000012ad8 arch_local_irq_restore+0x5c/0x80
[c000000001297d80] 000000005067a7b4 (unreliable)
[c000000001297da0] c00000000032e318 cpuidle_enter_state+0x1a0/0x344
[c000000001297df0] c0000000000a13f4 call_cpuidle+0x6c/0x74
[c000000001297e10] c0000000000a1804 do_idle+0x250/0x280
[c000000001297ea0] c0000000000a19c0 cpu_startup_entry+0x30/0x34
[c000000001297ed0] c00000000000f554 rest_init+0xd8/0xe4
[c000000001297f00] c0000000005e0c0c start_kernel+0x4f8/0x504
[c000000001297f90] c00000000000ad70 start_here_common+0x1c/0x52c

28:mon>

oohal, 2020-04-14:
> @stewartsmith Where do I write the powersave=off ? In the BMC?

Press `e` at the petitboot menu with the kernel you want to boot selected. That'll allow you to edit the parameters for the kernel that is going to be booted.

klauskiwi, 2020-04-14:
Since Yesenia is observing this issue on a Power8 system, and the workaround on this issue was created on HCODE, I asked Yesenia to open a separate issue to track this, so I'm closing this.

> @madscientist159 fix should be upstream now:
> 
> [open-power/hcode@cc0b781](https://github.com/open-power/hcode/commit/cc0b781abd67673240c208df7c77d2d37906057f)

closed 2020-04-14 by klauskiwi

svaidy, 2020-04-14:
> ```
> [    3.314037] powernv-cpufreq: cpufreq pstate min 0x63 nominal 0x63 max 0x0
> [    3.314068] powernv-cpufreq: Workload Optimized Frequency is enabled in the platform
> ```

This message is informational and does not indicate any error.  It says the cpu frequency scaling subsystem is inited and got all its inputs from firmware.  Further WoF is enabled in the platform.