open-power/op-build #2915: Skiroot dependencies for Secure & Trusted Boot

sammj, 2019-06-04:
Incoming changes for Secure & Trusted Boot on OpenPOWER platforms will have a few dependencies in the Skiroot image.
This may apparently include `efivar` and utilities from `efitools`.

There may be some porting work to be done here which the STB team will handle. More generally this will probably push up hard against the 16MB size limit for BOOTKERNEL or blow right past it.
The STB team will need to work with upstream op-build to work out
* What utilities are needed
* What their dependencies are
* What their size requirements are.

Possibly we'll have to look into increasing BOOTKERNEL size or potentially packing some tools as pb-plugins if possible.

naynajain, 2019-06-18:
We would want to submit the op-build patch for efivar asap, however we might have to do some changes to it in context of POWER. This might take us some time. 
But as Sam mentioned that it can push hard against the size limit. To get the confirmation on the size issue at the earliest, is it ok if we send the op-build patch now itself ? We will keep working on our POWER changes parallely and share the update once that is done.

I would like to get the feedback that if the suggested approach looks fine, or is there a better way ?

Thanks & Regards,
     - Nayna

sammj, 2019-06-18:
Sending through a patch now is the best way :) Then it can run through the pull-request CI and we can see how it handles it.

naynajain, 2019-06-19:
Thanks Sam !! We will try to send the patch asap.

Thanks & Regards,
          - Nayna

hemantbaxi, 2019-07-16:
Eric had tried building efivar into skiroot image. It seems it adds additional 300kb approximately.. He tested it and there was no complain on the size.
Also, it seems efivar is already available from buildroot. It just needs to be enabled via an openpower config - BR2_PACKAGE_EFIVAR=y. 
To try it now, we did via "op-build menuconfig". We are not very sure which config is the right one to be edited for the patch submission. Will it be openpower/configs/witherspoon_defconfig ?

naynajain, 2019-07-16:
Eric had tried building efivar into skiroot image. It seems it adds additional 300kb approximately.. He tested it and there was no complain on the size.
Also, it seems efivar is already available from buildroot. It just needs to be enabled via an openpower config - BR2_PACKAGE_EFIVAR=y.
To try it now, we did via "op-build menuconfig". We are not very sure which config is the right one to be edited for the patch submission. Will it be openpower/configs/witherspoon_defconfig ?

stewart-ibm, 2019-07-17:
Hemant Baxi <notifications@github.com> writes:
> Eric had tried building efivar into skiroot image. It seems it adds additional 300kb approximately.. He tested it and there was no complain on the size.
> Also, it seems efivar is already available [...]

All platforms that will support secure boot, which means all POWER9 ones.

-- 
Stewart Smith
OPAL Architect, IBM.