open-power/op-build #2973: op-build master and v2.3 fails to build on rhel 8.1 beta

klauskiwi, 2019-07-04:
When trying to op-build opal_defconfig on a ppc64el RHEL 8.1 beta, I'm getting the following:
`#
# configuration written to .config
#
>>> linux 5.0.7 Configuring
mkdir -p "/home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version"
# Add package name
echo -n "       linux-" > /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt
# If site local
# Add site local and user, local commit, if local is dirty
# Else not local
# Add package version, extraversion if linux, op-build is dirty, op-build patches
if [ "https" == "local" ]; then echo -n "site_local-" >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt; whoami | xargs echo -n >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt; echo -n "-" >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt; cd ""https://cdn.kernel.org/pub"/linux/kernel/v5.x"; (git describe --always --dirty || echo "unknown") |sed -e 's/linux-//' | xargs echo -n >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt; else [ `echo -n 5.0.7 | wc -c` == "40" ] && (echo -n 5.0.7 | sed "s/^\([0-9a-f]\{7\}\).*/\1/;s/linux-//;" >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt) || echo -n 5.0.7 | sed -e 's/linux-//' >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt; if [ "LINUX" == "LINUX" ]; then if ls /home/klausk/op-build/output/build/linux-5.0.7/Makefile 1>/dev/null; then head /home/klausk/op-build/output/build/linux-5.0.7/Makefile | grep EXTRAVERSION | cut -d ' ' -f 3 | xargs echo -n >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt; fi; fi; if [ -f /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.patch.txt ]; then echo -n "-p" >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt; cat /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.patch.txt >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt; fi fi
# Add new line to version.txt
echo "" >> /home/klausk/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/linux.version.txt;
>>> linux 5.0.7 Building
PATH="/home/klausk/op-build/output/host/bin:/home/klausk/op-build/output/host/sbin:/home/klausk/.local/bin:/home/klausk/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin" BR_BINARIES_DIR=/home/klausk/op-build/output/images /usr/bin/make -j129 HOSTCC="/usr/bin/gcc -O2 -I/home/klausk/op-build/output/host/include -L/home/klausk/op-build/output/host/lib -Wl,-rpath,/home/klausk/op-build/output/host/lib" ARCH=powerpc INSTALL_MOD_PATH=/home/klausk/op-build/output/target CROSS_COMPILE="/home/klausk/op-build/output/host/bin/powerpc64le-buildroot-linux-gnu-" DEPMOD=/home/klausk/op-build/output/host/sbin/depmod INSTALL_MOD_STRIP=1 -C /home/klausk/op-build/output/build/linux-5.0.7 zImage.epapr
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/zconf.lex.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
  SYSTBL  arch/powerpc/include/generated/asm/syscall_table_32.h
  SYSTBL  arch/powerpc/include/generated/asm/syscall_table_64.h
  SYSTBL  arch/powerpc/include/generated/asm/syscall_table_c32.h
  SYSTBL  arch/powerpc/include/generated/asm/syscall_table_spu.h
  SYSHDR  arch/powerpc/include/generated/uapi/asm/unistd_32.h
  SYSHDR  arch/powerpc/include/generated/uapi/asm/unistd_64.h
  WRAP    arch/powerpc/include/generated/uapi/asm/param.h
  WRAP    arch/powerpc/include/generated/uapi/asm/poll.h
  WRAP    arch/powerpc/include/generated/uapi/asm/resource.h
  WRAP    arch/powerpc/include/generated/uapi/asm/siginfo.h
  WRAP    arch/powerpc/include/generated/uapi/asm/sockios.h
  WRAP    arch/powerpc/include/generated/uapi/asm/statfs.h
  UPD     include/config/kernel.release
  UPD     include/generated/uapi/linux/version.h
  UPD     include/generated/utsrelease.h
  WRAP    arch/powerpc/include/generated/asm/div64.h
  WRAP    arch/powerpc/include/generated/asm/export.h
  WRAP    arch/powerpc/include/generated/asm/irq_regs.h
  WRAP    arch/powerpc/include/generated/asm/irq_work.h
  WRAP    arch/powerpc/include/generated/asm/local64.h
  WRAP    arch/powerpc/include/generated/asm/mcs_spinlock.h
  WRAP    arch/powerpc/include/generated/asm/preempt.h
  WRAP    arch/powerpc/include/generated/asm/rwsem.h
  WRAP    arch/powerpc/include/generated/asm/vtime.h
  WRAP    arch/powerpc/include/generated/asm/msi.h
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  LEX     scripts/dtc/dtc-lexer.lex.c
  YACC    scripts/dtc/dtc-parser.tab.h
  YACC    scripts/dtc/dtc-parser.tab.c
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTLD  scripts/dtc/dtc
  HOSTCC  scripts/bin2c
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/pnmtologo
  HOSTCC  scripts/conmakehash
  HOSTCC  scripts/sortextable
  HOSTCC  scripts/asn1_compiler
  HOSTCC  scripts/sign-file
  HOSTCC  scripts/extract-cert
scripts/extract-cert.c:21:10: fatal error: openssl/bio.h: No such file or directory
 #include <openssl/bio.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.host:90: scripts/extract-cert] Error 1
make[3]: *** Waiting for unfinished jobs....
scripts/sign-file.c:25:10: fatal error: openssl/opensslv.h: No such file or directory
 #include <openssl/opensslv.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.host:90: scripts/sign-file] Error 1
make[2]: *** [Makefile:1080: scripts] Error 2
make[1]: *** [package/pkg-generic.mk:241: /home/klausk/op-build/output/build/linux-5.0.7/.stamp_built] Error 2
make: *** [Makefile:84: _all] Error 2
make: Leaving directory '/home/klausk/op-build/buildroot'
[klausk@faraday op-build]$ cat /etc/
Display all 249 possibilities? (y or n)
[klausk@faraday op-build]$ cat /etc/redhat-
redhat-access-insights/ redhat-release
[klausk@faraday op-build]$ cat /etc/redhat-release
Red Hat Enterprise Linux release 8.1 Beta (Ootpa)
`
The above happened on when trying to build op-build tag v2.3 after trying HEAD with similar results.

klauskiwi, 2019-07-04:
Build succeeded after installing openssl-devel and dtc.. if we had an explicit RedHat-centric documentation, those should probably be listed there, but for now I'll just close this

stewart-ibm, 2019-07-04:
Klaus Heinrich Kiwi <notifications@github.com> writes:
> Build succeeded after installing openssl-devel and dtc.. if we had an
> explicit RedHat-centric documentation, those should probably be listed
> there, but for now I'll just close this

Patches welcome :)

-- 
Stewart Smith
OPAL Architect, IBM.

shenki, 2019-07-04:
I will push a fix for this.

klauskiwi, 2019-07-04:
Thanks for re-opening and providing a patch. Since I'm not able to install Fedora30 at this point, I'm continuing to do some experiments with the default RHEL 8.1 beta installation.

So for RHEL 8.1 beta, in addition to `openssl-devel` and `dtc`, we also need to add `rsync` and "perl(ExtUtils::MakeMaker)" to the package list and adjust `python` for `python2`.  After installing the packages, we need to adjust the default python interpreter using `sudo update-alternatives --set python /usr/bin/python2` as well.

I'll try to check if this is also true for Fedora30, so I'd suggest holding on pushing a fix before we know the complete picture - sometimes we'll need a new section for RHEL after all.

 -Klaus

closed 2019-07-04 by stewart-ibm