open-power/op-build #2988: SLW: timer stuck on palmetto

shenki, 2019-07-10:
```SLW: timer stuck, falling back to OPAL pollers. You will likely have slower I2C and may have experienced increased jitter.
SLW: Stuck with odd generation !
SLW: Register state:
SLW:  e0000 = 0e01000800000acc
SLW:  e0001 = 0000ffffffffffff
SLW:  e0002 = 0000000000000000
SLW:  e0003 = 00bff00000000000
SLW:  e0004 = 0000000000000000
SLW:  e0005 = 0000000000000000
SLW:  e0006 = 00000000003405df
SLW:  e0007 = 000000000000000a
SLW:  e0008 = 0000000800000000
SLW:  e0009 = 0000000000000000
SLW:  e000a = 0000004200000000
SLW:  e000b = 0000000000000000
SLW:  e000c = 0000000000000000
SLW:  e000d = 0000000000000000
SLW:  e000e = 0000000000000000
SLW:  e000f = 0008015c00000000
SLW:  e0010 = 000800000acc0000
SLW:  e0011 = 0000000000000000
SLW:  e0012 = 0000000000000000
SLW:  e0013 = 0000000000000000
SLW:  e0014 = 0000000000000904
SLW:  e0015 = 0000000300000000
SLW:  e0016 = 0000000280002000
SLW:  e0017 = c200510500000000
SLW:  e0018 = b400000300000000
SLW:  e0019 = 0000000000000000
SLW:  5001c = a800fb0100000000
SLW:  50038 = 0000000100000000
SLW:  50039 = 4004002600000000
SLW:  5003a = 0000000000000000
SLW:  5003b = 806c000000000000
```

```
open-power-palmetto-v2.4-rc1-3-g38677982
	buildroot-2019.05-6-g0d430a2acb
	skiboot-v6.4-rc1
	hostboot-p8-27dda29-p580ec27
	occ-p8-a2856b7
	linux-5.1.16-openpower1-p80e270e
	petitboot-v1.10.4
	machine-xml-19aa4d3
	hostboot-binaries-hw062819a.940
	capp-ucode-p9-dd2-v4
```

shenki, 2019-07-10:
I also saw this with an older build:
```
open-power-palmetto-v2.3-rc2-60-g3c8d15f0
	buildroot-2019.02.2-17-g93b841d204
	skiboot-v6.3-rc2
	hostboot-p8-0b35de9-p580ec27
	occ-p8-a2856b7
	linux-5.0.9-openpower1-p380bfa0
	petitboot-v1.10.3
	machine-xml-19aa4d3
	hostboot-binaries-hw043019a.940
	capp-ucode-p9-dd2-v4
```
```
SLW: timer stuck, falling back to OPAL pollers. You will likely have slower I2C and may have experienced increased jitter.
SLW: Stuck with odd generation !
SLW: Register state:
SLW:  e0000 = 0201000800000ae4
SLW:  e0001 = 8000ffffffffffff
SLW:  e0002 = 0000000000000000
SLW:  e0003 = 00bff00000000000
SLW:  e0004 = 0000000000000000
SLW:  e0005 = 0000000000000000
SLW:  e0006 = 0000000000231dcf
SLW:  e0007 = 000000000000000a
SLW:  e0008 = 0000000800000000
SLW:  e0009 = 0000000000000000
SLW:  e000a = 0000000000000000
SLW:  e000b = 0000000000000000
SLW:  e000c = 0000000000000000
SLW:  e000d = 0000000000000000
SLW:  e000e = 0000000000000000
SLW:  e000f = 0005003a70000000
SLW:  e0010 = 000800000ad80001
SLW:  e0011 = 0000000000000000
SLW:  e0012 = 0000000000000000
SLW:  e0013 = 0000000000000000
SLW:  e0014 = 0000000070000904
SLW:  e0015 = 0000000300000000
SLW:  e0016 = 0000000280002000
SLW:  e0017 = c200510500000000
SLW:  e0018 = b400000300000000
SLW:  e0019 = 0000000000000000
```

shenki, 2019-07-10:
The differences between the old firmware (-) and the new (+):
```
 SLW: timer stuck, falling back to OPAL pollers. You will likely have slower I2C and may have experienced increased jitter.
 SLW: Stuck with odd generation !
 SLW: Register state:
-SLW:  e0000 = 0201000800000ae4
-SLW:  e0001 = 8000ffffffffffff
+SLW:  e0000 = 0e01000800000acc
+SLW:  e0001 = 0000ffffffffffff
 SLW:  e0002 = 0000000000000000
 SLW:  e0003 = 00bff00000000000
 SLW:  e0004 = 0000000000000000
 SLW:  e0005 = 0000000000000000
-SLW:  e0006 = 0000000000231dcf
+SLW:  e0006 = 00000000003405df
 SLW:  e0007 = 000000000000000a
 SLW:  e0008 = 0000000800000000
 SLW:  e0009 = 0000000000000000
-SLW:  e000a = 0000000000000000
+SLW:  e000a = 0000004200000000
 SLW:  e000b = 0000000000000000
 SLW:  e000c = 0000000000000000
 SLW:  e000d = 0000000000000000
 SLW:  e000e = 0000000000000000
-SLW:  e000f = 0005003a70000000
-SLW:  e0010 = 000800000ad80001
+SLW:  e000f = 0008015c00000000
+SLW:  e0010 = 000800000acc0000
 SLW:  e0011 = 0000000000000000
 SLW:  e0012 = 0000000000000000
 SLW:  e0013 = 0000000000000000
-SLW:  e0014 = 0000000070000904
+SLW:  e0014 = 0000000000000904
 SLW:  e0015 = 0000000300000000
 SLW:  e0016 = 0000000280002000
 SLW:  e0017 = c200510500000000
 SLW:  e0018 = b400000300000000
 SLW:  e0019 = 0000000000000000
+SLW:  5001c = a800fb0100000000
+SLW:  50038 = 0000000100000000
+SLW:  50039 = 4004002600000000
+SLW:  5003a = 0000000000000000
+SLW:  5003b = 806c000000000000
```

dcrowell77, 2019-07-10:
Which timer is this?  I thought you went through the SBE, but the traces say SLW (which is the old name for the cme/gpe/hcode/pm complex).

stewart-ibm, 2019-07-16:
Dan <notifications@github.com> writes:
> Which timer is this?  I thought you went through the SBE, but the
> traces say SLW (which is the old name for the cme/gpe/hcode/pm
> complex).

A bunch of our code still says 'slw' even on P9 (it's even baked into
some OPAL call names).

Although palmetto is p8, so I guess SLW is the correct one there.

-- 
Stewart Smith
OPAL Architect, IBM.

hegdevasant, 2019-07-30:
For P8 system we used SLW based timer.

For P9, we are using SBE provided one shot timer chip-op.

-Vasant

hegdevasant, 2019-07-30:
Looks like this is SLW issue... OPAL didn't get response on time. 

-Vasant

stewartsmith, 2020-11-09:
Since POWER8 and palmetto support was removed in 40f804058414ddf4518ac07866bb58af74db20a2, this can probably be closed now.

closed 2020-11-09 by dcrowell77