open-power/op-build #2989: opal-prd fails to start on palmetto

shenki, 2019-07-10:
```
CTRL: Starting PRD daemon
I2C: Found Chip: 00000000 engine 1 port 2
I2C: Found Chip: 00000000 engine 0 port 0
I2C: Found Chip: 00000000 engine 0 port 1
I2C: Found Chip: 80000001 engine 0 port 0
CTRL: Listening on control socket /run/opal-prd-control
FW: 5 PRD ranges, instances assigned by firmware
FW:  0000000ffd450000-0000000ffd6dffff ibm,hbrt-code-image
FW:  0000000ffd6e0000-0000000ffd6fffff ibm,hbrt-target-image
FW:  0000000ffd700000-0000000ffd7fffff ibm,hbrt-vpd-image
FW:  0000000ffd800000-0000000ffdbfffff ibm,homer-image [0]
FW:  0000000fff800000-0000000fffffffff ibm,occ-common-area
FW: Chip init
FW: Chip 0x0
IMAGE: hbrt map at 0x7fff9ed70000, size 0x290000
HBRT: calling hservices_init
IMAGE: code address: 0x7fff9ed70000
IMAGE: Bad signature for ibm,hbrt-code-image! exiting
HBRT: Can't initialise HBRT
CTRL: stopping PRD daemon
```

```
open-power-palmetto-v2.4-rc1-3-g38677982
	buildroot-2019.05-6-g0d430a2acb
	skiboot-v6.4-rc1
	hostboot-p8-27dda29-p580ec27
	occ-p8-a2856b7
	linux-5.1.16-openpower1-p80e270e
	petitboot-v1.10.4
	machine-xml-19aa4d3
	hostboot-binaries-hw062819a.940
	capp-ucode-p9-dd2-v4
```

```
$ opal-prd --version
opal-prd opal-prd-6.2
```

Host is Debian 4.19 ppc64le.

dcrowell77, 2019-07-10:
Never seen that error before.  I wasn't aware that opal-prd/skiboot/linux did any verification of the HBRT image.

stewart-ibm, 2019-07-16:
Dan <notifications@github.com> writes:
> Never seen that error before.  I wasn't aware that
> opal-prd/skiboot/linux did any verification of the HBRT image.

We check the magic number at least, just to ensure a bare amount of
sanity.

It's literally just this code:

        if (memcmp(code, "HBRTVERS", 8) != 0) {
                pr_log(LOG_ERR, "IMAGE: Bad signature for "
                                "ibm,hbrt-code-image! exiting");
                return -1;
        }

-- 
Stewart Smith
OPAL Architect, IBM.

stewartsmith, 2020-11-09:
Since POWER8 and palmetto support was removed in 40f804058414ddf4518ac07866bb58af74db20a2, this can probably be closed now.

closed 2020-11-09 by dcrowell77