open-power/op-build #3146: palmetto fails in istep 16.1

shenki, 2019-09-16:
Using op-build master on Palmetto:

```
85.66137|ISTEP 16. 1 - host_activate_master
 85.83588|================================================
 85.84158|Error reported by hwpf (0x0900)
 85.84728|  HW Procedure generated error. See User Data.
 85.84728|  ModuleId   0x02 MOD_HWP_RC_TO_ERRL
 85.84728|  ReasonCode 0x090f RC_HWP_GENERATED_ERROR
 85.84729|  UserData1  RC value from HWP : 0x00000000007c328a
 85.84730|  UserData2  <unused> : 0x0000000000000000
 85.84730|User Data Section 0, type UD
 85.84730|  Subsection type 0x01
 85.84731|  ComponentId hwpf (0x0900)
 85.84731|User Data Section 1, type UD
 85.84732|  Subsection type 0x02
 85.84732|  ComponentId hwpf (0x0900)
 85.84732|User Data Section 2, type UD
 85.84733|  Subsection type 0x02
 85.84733|  ComponentId hwpf (0x0900)
 85.84733|User Data Section 3, type UD
 85.84734|  Subsection type 0x02
 85.84734|  ComponentId hwpf (0x0900)
 85.84735|User Data Section 4, type UD
 85.84735|  Subsection type 0x06
 85.84735|  ComponentId errl (0x0100)
 85.84736|  CALLOUT
 85.84736|  PROCEDURE ERROR
 85.84736|  Procedure: 85
 85.84737|User Data Section 5, type UD
 85.84737|  Subsection type 0x04
 85.84737|  ComponentId errl (0x0100)
 85.84738|User Data Section 6, type UD
 85.84738|  Subsection type 0x06
 85.84738|  ComponentId errl (0x0100)
 85.84739|  CALLOUT
 85.84739|  HW CALLOUT
 85.84739|  Reporting CPU ID: 24
 85.84740|  Called out entity:
 85.84740|User Data Section 7, type UD
 85.84740|  Subsection type 0x15
 85.84741|  ComponentId hb-trace (0x3100)
 85.84741|User Data Section 8, type UD
 85.84741|  Subsection type 0x15
 85.84742|  ComponentId hb-trace (0x3100)
 85.84742|User Data Section 9, type UD
 85.84743|  Subsection type 0x15
 85.84743|  ComponentId hb-trace (0x3100)
 85.84743|User Data Section 10, type UD
 85.84744|  Subsection type 0x02
 85.84744|  ComponentId errl (0x0100)
 85.84744|  TARGET
 85.84745|  HUID: 00060002
 85.84746|  Phys path: Physical:/Sys0/Node0/Proc0/EX3
 85.84747|  Affinity path: Logical:/Sys0/Node0/Proc0/EX3
 85.84747|User Data Section 11, type UD
 85.84748|  Subsection type 0x15
 85.84748|  ComponentId hb-trace (0x3100)
 85.84749|User Data Section 12, type UD
 85.84749|  Subsection type 0x03
 85.84749|  ComponentId errl (0x0100)
 85.84750|User Data Section 13, type UD
 85.84750|  Subsection type 0x01
 85.84750|  ComponentId errl (0x0100)
 85.84751|  STRING
 85.84751|  Hostboot Build ID: 
 85.84751|User Data Section 14, type UD
 85.84752|  Subsection type 0x04
 85.84752|  ComponentId errl (0x0100)
 85.84753|================================================
 86.54981|System shutting down with error status 0x900000BD
 86.54982|================================================
 86.54983|Error reported by istep (0x1700)
 86.83506|  IStep failed, see other log(s) with the same PLID for reason.
 86.83506|  ModuleId   0x01 ISTEP_REPORTING_ERROR
 86.83506|  ReasonCode 0x1701 ISTEP_FAILURE
 86.83507|  UserData1  eid of first error : 0x900000bd0000090f
 86.83507|  UserData2  Reason code of first error : 0x0000000100001001
 86.83507|User Data Section 0, type UD
 86.83508|  Subsection type 0x03
 86.83508|  ComponentId hwpf (0x0900)
 86.83508|User Data Section 1, type UD
 86.83509|  Subsection type 0x01
 86.83509|  ComponentId errl (0x0100)
 86.83509|  STRING
 86.83509|  host_activate_master
 86.83510|User Data Section 2, type UD
 86.83510|  Subsection type 0x06
 86.83510|  ComponentId errl (0x0100)
 86.83510|  CALLOUT
 86.83511|  PROCEDURE ERROR
 86.83511|  Procedure: 85
 86.83511|User Data Section 3, type UD
 86.83511|  Subsection type 0x03
 86.83512|  ComponentId errl (0x0100)
 86.83512|User Data Section 4, type UD
 86.83512|  Subsection type 0x01
 86.83512|  ComponentId errl (0x0100)
 86.83513|  STRING
 86.83513|  Hostboot Build ID: 
 86.83513|================================================
```

dcrowell77, 2019-09-16:
85.84729|  UserData1  RC value from HWP : 0x00000000007c328a

obj/genfiles/fapiHwpReturnCodes.H:    RC_PROC_PREP_MASTER_WINKLE_SBE_NOT_RUNNING = 0x7c328a,

That indicates that the SBE isn't running when we are ready to do the master winkle procedures in Hostboot.  I'd be surprised if there have been any SBE changes recently in the P8 code.  Is it a 1-time failure or consistent?

stewartsmith, 2020-11-08:
Since POWER8 and palmetto support was removed in 40f804058414ddf4518ac07866bb58af74db20a2, this can probably be closed now.

closed 2020-11-09 by dcrowell77