open-power/op-build #3274: Client Issue:  Intel XL710 Ethernet TX Queue Lockups under heavy workload

djmccoyibm, 2019-11-01:
This client purchased a P9 openpower system for the purpose of running Linux (RHEL 7.6ALT) with Spectrum Scale NFS workloads on it. They have installed an XL710 10GBE Fiber Ethernet adapter, and are having troubles with TX lockups where the claim is the adapter will not accept new TX packets and has to be reset by the i40e driver to recover. I've attached relevant log snippets below followed by configuration information.

From the Linux operating system perspective, the logs were examined by a Red Hat network specialist, and this is what he said which points us towards further investigation from the hardware side:
--
The concern is the tx timeout on the NICs such as:
tx_timeout: VSI_seid: 396, Q 31, NTC: 0x1cb, HWB: 0x1cb, NTU: 0x1f5, TAIL: 0x1f5, INT: 0x1</i>
<i> tx_timeout recovery level 2, hung_queue 31
tx_timeout: VSI_seid: 396, Q 43, NTC: 0x113, HWB: 0x113, NTU: 0x13e, TAIL: 0x13e, INT: 0x1</i>
<i> tx_timeout recovery level 1, hung_queue 43

This is the driver detecting a problem with the NIC no longer accepting transmit, so the driver resets the NIC to get it back into a working state.
The resets appear to work, and are the reason the earlier regular init messages are logged.

The tx_timeout messages don't shed a great deal of light on the problem to us. Reading the driver source and XL710 series datasheet, the "VSI_seid" is an internal hardware identifier, and "INT: 0x01" means interrupts are enabled, the rest are just housekeeping for queue processing.

The driver goes through various levels of reset. First trying just the PF, then what it calls "core", then what it calls "global", with the last being the most severe sort of reset.

SW teams claim this is a hardware issue, so please have the owner of the hardware pursue this via the hardware support channel.  The fix may indeed be a driver change, but we need to know *why* the hardware is in a hung state to know what to fix.

As we don't know the cause, suggesting a workaround is difficult. You may try disabling all transmit offloading with "ethtool --offload" to see if that causes the hardware to behave differently.
--

The network configuration is as follows (note: They are using all four ports, configured the same from a physical perspective):

Settings for enP48p1s0f0:
Supported ports: [ FIBRE ]
Supported link modes: 1000baseT/Full
10000baseT/Full
Supported pause frame use: Symmetric
Supports auto-negotiation: No
Supported FEC modes: Not reported
Advertised link modes: 1000baseT/Full
10000baseT/Full
Advertised pause frame use: No
Advertised auto-negotiation: No
Advertised FEC modes: Not reported
Speed: 10000Mb/s
Duplex: Full
Port: FIBRE
PHYAD: 0
Transceiver: internal
Auto-negotiation: off
Supports Wake-on: g
Wake-on: g
Current message level: 0x00000007 (7)
drv probe link
Link detected: yes

driver: i40e
version: 2.1.14-k
firmware-version: 6.01 0x800036e4 1.1861.0
expansion-rom-version:
bus-info: 0030:01:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes

The client has a bond set up over enP48p1s0f0 and f1, and another bond set up over enP48p1s0f2 and f3. Over each bond is a tagged 802.11ad VLAN. One VLAN is toward Spectrum Scale I/O nodes, the other is toward client nodes. The primary protocol toward client nodes is NFS. Toward the I/O nodes is RDMA over Converged Ethernet.

As you can see, the cards are fairly up to date on firmware, but we are curious if another FW level is available above that level that may have a fix affecting the issues above. We understand that there is an issue with update programs for this card on ppc64le, and we'll deal with that if/when we need to. If this doesn't seem know, what data would you need to pursue further with the IO card? We do have sosreport data from the affected ppc64le Linux systems that could be shared if/when needed.

I also learned more about the driver in use.  The driver for i40e as reported by the Linux kernel may either be the level reported by the kmod-i40e rpm (if installed), or whatever the last update to the Linux Kernel had.  The kernel may not report the level of fixes actually installed.  We have been requested by the HW provider to update the driver to 2.4.10 level from where it currently is, however, that level is not available through RHEL and RHEL has said that they would not support gooing above the level in the kernel / rpms right now, and would expect that support to come from the HW Mfg if the driver is to be above the kernel level.

stewartsmith, 2019-11-01:
This seems like an OS level issue and not something to do with the system firmware (which is what op-build project is about).

djmccoyibm, 2019-11-01:
Yeah.  Found a better place.  Sooooo sorry!

closed 2019-11-01 by djmccoyibm