open-power/op-build #3330: Doesn't build on Fedora 31

stewartsmith, 2019-12-10:
Looks like SBE doesn't build with /usr/bin/python being python 3.x like it is in fedora 31.

```
  File "/home/stewart/op-build/output/build/sbe-8b06b5c1bd307c0052f25a79da50fbecf72093b3/src/build/security/securityRegListGen.py", line 64
    -v, --verbose            enable verbose traces"
                                                  ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("usage: p9_security_list_gen.py [-h] [-f <security_list_path>] [-0 <output directory] [i] [-d] [-v]\n\
arguments:\n\
-h, --help               show this help message and exit\n\
-f, --file               path to the security list csv file\n\
-o, --output             output directory\n\
-w, --whitelist          print whitelist read from csv\n\
-b, --blacklist          print blacklist read from csv\n\
-g, --greylist           print greylist read from csv\n\
-i, --info               get version info of the security list\n\
-d, --debug              enable debug traces\n\
-v, --verbose            enable verbose traces")?
  File "/home/stewart/op-build/output/build/sbe-8b06b5c1bd307c0052f25a79da50fbecf72093b3/src/build/security/securityRegListGen.py", line 64
    -v, --verbose            enable verbose traces"
                                                  ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("usage: p9_security_list_gen.py [-h] [-f <security_list_path>] [-0 <output directory] [i] [-d] [-v]\n\
arguments:\n\
-h, --help               show this help message and exit\n\
-f, --file               path to the security list csv file\n\
-o, --output             output directory\n\
-w, --whitelist          print whitelist read from csv\n\
-b, --blacklist          print blacklist read from csv\n\
-g, --greylist           print greylist read from csv\n\
-i, --info               get version info of the security list\n\
-d, --debug              enable debug traces\n\
-v, --verbose            enable verbose traces")?
make[4]: *** [Makefile:324: security] Error 1
make[3]: *** [Makefile:179: all] Error 2
make[2]: *** [Makefile:36: DD2] Error 2
make[2]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:324: security] Error 1
make[3]: *** [Makefile:179: all] Error 2
make[2]: *** [Makefile:39: axone] Error 2
make[1]: *** [package/pkg-generic.mk:241: /home/stewart/op-build/output/build/sbe-8b06b5c1bd307c0052f25a79da50fbecf72093b3/.stamp_built] Error 2
```

dcrowell77, 2019-12-10:
@rajadas2 - Take a look

stewartsmith, 2019-12-10:
There’s an old PR to migrate the code to Python 3 from a community member which has been ignored, although the easy (albeit temporary) solution is to just explicitly call python2.

stewartsmith, 2020-02-25:
I sent this PR https://github.com/open-power/sbe/pull/17 which hasn't received any attention for 2.5 months.

klauskiwi, 2020-07-07:
Looks like this can finally be closed. Upstream SBE builds fine with python pointing at python3

closed 2020-07-07 by klauskiwi