open-power/op-build #3484: blackbird_defconfig results in 404 on download of machine-xml 17e9e84d504582c881e480b82fe1cf115312130f Download

ldr, 2020-02-27:
Hi,

It's the first time I'm using buildroot, so maybe I'm doing something wrong, but when using the blackbird_defconfig, I get the following error:
```>> machine-xml 17e9e84d504582c881e480b82fe1cf115312130f Downloading
--2020-02-28 00:10:53--  https://github.com/open-power/blackbird-xml/archive/17e9e84d504582c881e480b82fe1cf115312130f/machine-xml-17e9e84d504582c881e480b82fe1cf115312130f.tar.gz
Resolving github.com (github.com)... 140.82.118.4
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-28 00:10:53 ERROR 404: Not Found.

--2020-02-28 00:10:53--  http://sources.buildroot.net/machine-xml/machine-xml-17e9e84d504582c881e480b82fe1cf115312130f.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.0.37, 104.26.1.37, 2606:4700:20::681a:25, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.0.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-28 00:10:53 ERROR 404: Not Found.
>> machine-xml 17e9e84d504582c881e480b82fe1cf115312130f Downloading
--2020-02-28 00:10:53--  https://github.com/open-power/blackbird-xml/archive/17e9e84d504582c881e480b82fe1cf115312130f/machine-xml-17e9e84d504582c881e480b82fe1cf115312130f.tar.gz
Resolving github.com (github.com)... 140.82.118.4
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-28 00:10:53 ERROR 404: Not Found.

--2020-02-28 00:10:53--  http://sources.buildroot.net/machine-xml/machine-xml-17e9e84d504582c881e480b82fe1cf115312130f.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.0.37, 104.26.1.37, 2606:4700:20::681a:25, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.0.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-28 00:10:53 ERROR 404: Not Found.

--2020-02-28 00:10:53--  http://sources.buildroot.net/machine-xml-17e9e84d504582c881e480b82fe1cf115312130f.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 104.26.0.37, 2606:4700:20::681a:125, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-28 00:10:53 ERROR 404: Not Found.

make: *** [package/pkg-generic.mk:146: /home/ldr/src/op-build/output/build/machine-xml-17e9e84d504582c881e480b82fe1cf115312130f/.stamp_downloaded] Error 1

--2020-02-28 00:10:53--  http://sources.buildroot.net/machine-xml-17e9e84d504582c881e480b82fe1cf115312130f.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 104.26.0.37, 2606:4700:20::681a:125, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-28 00:10:53 ERROR 404: Not Found.

make: *** [package/pkg-generic.mk:146: /home/ldr/src/op-build/output/build/machine-xml-17e9e84d504582c881e480b82fe1cf115312130f/.stamp_downloaded] Error 1
```
Is this something I can fix myself ? or is this file correct and is it actually missing ?

Thanks & best regards,

Leon

stewartsmith, 2020-02-28:
It’s not. You have to clone my repo and do the site method and location to use a local clone.

Nobody answered my request for a repo back in December: https://lists.ozlabs.org/pipermail/openpower-firmware/2019-December/000439.html

ldr, 2020-02-29:
Hello Stewart, thanks for your reply! Found some of your videos and then your blog describing your fork and changes, really interesting, it will certainly help me progress getting buildroot running ! Thanks again :)

klauskiwi, 2020-03-02:
Hi Stewart

On 2/27/2020 10:42 PM, Stewart Smith wrote:
> It’s not. You have to clone my repo and do the site method and location 
> to use a local clone.
> 
> Nobody answered my request for a repo back in December: 
> https://lists.ozlabs.org/pipermail/openpower-firmware/2019-December/000439.html

I apologize for the delay. I routed your e-mail internally back in 
December but forgot to keep poking it. I'm back at it again.

Thanks,

  -Klaus

-- 
Klaus Heinrich Kiwi <klaus@linux.vnet.ibm.com>

stewartsmith, 2020-04-01:
The repo has been created, and a test build I just did shows that this issue is fixed and should be able to be closed now.
@ldr - are you able to retry to confirm?

ldr, 2020-04-02:
Nice, the build finished :) Thanks a lot @klauskiwi and @stewartsmith !!

closed 2020-04-02 by ldr