open-power/op-build #3678: [swift] op-build legal-info fails

hegdevasant, 2020-06-08:
Looks like we are not downloading OCMB LICENSE file . Hence op-build legal-info fails.

18:09:12 [7m>>> ocmb-explorer-fw v4.0 Collecting legal info[27m
18:09:12 WARNING: no hash file for LICENSE
18:09:12 cp: cannot stat '/jenkins/jenkins/workspace/op-build/op-build-master/PLATFORM/swift/output-ubuntu1804/swift_defconfig/build/ocmb-explorer-fw-v4.0/LICENSE': No such file or directory
18:09:12 /jenkins/jenkins/workspace/op-build/op-build-master/PLATFORM/swift/openpower/package/ocmb-explorer-fw/ocmb_explorer_fw.mk:27: recipe for target 'ocmb-explorer-fw-legal-info' failed
18:09:12 make: *** [ocmb-explorer-fw-legal-info] Error 1
18:09:12 make: Leaving directory '/jenkins/jenkins/workspace/op-build/op-build-master/PLATFORM/swift/buildroot'
18:09:13 Build step 'Execute shell' marked build as failure
18:09:13 New run name is '#94_96c65638'

klauskiwi, 2020-06-17:
@wghoffa @varada28 I believe we will need your help here, as apparently the ocmb-explorer-fw releases are totally non-standard...

op-build is expecting that we download a source-code tarbal, and inside it we'll find some license file (specified by `OCMB_EXPLORER_FW_LICENSE_FILES`) as well as source-code and scripts/Makefiles to build the binary..

Even if we're distributing the ocmb firmware file as binary only (and I'd like to understand why), I think we could at least make a tarball/zip file containing the binary AND the license file for which we will point to and collect during `op-build legal-info`.

klauskiwi, 2020-06-18:
Tentative fix here https://github.com/open-power/op-build/pull/3709

klauskiwi, 2020-07-02:
#3709 fixes this

closed 2020-07-02 by klauskiwi