open-power/op-build #4070: Fails to IPL Blackbird system

stewartsmith, 2021-04-11:
Current op-build master, to be exact, this (the dirty is a change to the "build in a container script")
```
open-power-blackbird-v2.6-185-gd17829dd-dirty
	buildroot-2020.08-293-ga84aaaeacb
	skiboot-v6.7-34-g3d3c2b186
	hostboot-0d0700f-pe7e80e1
	occ-16131c3
	linux-5.4.107-openpower1-pf974b06
	petitboot-v1.12
	machine-xml-e782e30
	hostboot-binaries-hw101520a.opmst
	capp-ucode-p9-dd2-v4
	sbe-5571071
	hcode-hw040821a.opmst
```
fails to IPL a Raptor Blackbird which has been working fine with v2.5 based firmware (specifically https://www.flamingspork.com/blog/2020/05/25/op-build-v2-5-firmware-for-the-raptor-blackbird/ )

IPL failure is:

```
90.08424|ISTEP 21. 1 - host_runtime_setup
 90.58812|htmgt|OCCs are not active (rc=0x2625). Attempting OCC Reset
 90.60196|================================================
 90.60494|Error reported by hwpf (0x0900) EID 0x900000D4
 90.60644|  HW Procedure generated error. See User Data.
 90.60645|  ModuleId   0x01 MOD_FAPI2_RC_TO_ERRL
 90.60645|  ReasonCode 0x090f RC_HWP_GENERATED_ERROR
 90.60646|  UserData1  RC value from HWP : 0x0000000000b12b5b
 90.60647|  UserData2  <unused> : 0x0000000000000000
 90.60648|------------------------------------------------
 90.63866|  HwpReturnCode              : RC_SPCWKUP_EX_TIMEOUT
 90.63867|  HWP Error description      : Special wakeup to EX chiplet timed out.
 90.63868|------------------------------------------------
 90.64236|  HwpReturnCode              : RC_PPE_BASE_ADDR_XIXCR
 90.64392|  FFDC                       : PPE_BASE_ADDR
 90.64394|     00000000 13012410
 90.66786|------------------------------------------------
 90.66789|  HwpReturnCode              : RC_PPE_STATE_DATA_XIR
 90.66790|  FFDC                       : REG_XSR
 90.66791|     C9001400
 90.66795|------------------------------------------------
 90.66797|  HwpReturnCode              : RC_PPE_STATE_DATA_XIR
 90.66798|  FFDC                       : REG_IAR
 90.66799|     FFFF884C
 90.66803|------------------------------------------------
 90.66805|  HwpReturnCode              : RC_PPE_STATE_DATA_XIR
 90.66806|  FFDC                       : REG_IR
 90.66807|     7FFCF808
 90.66811|------------------------------------------------
 90.66813|  HwpReturnCode              : RC_PPE_STATE_DATA_XIR
 90.66814|  FFDC                       : REG_EDR
 90.66815|     00000000
 90.66818|------------------------------------------------
 90.66821|  HwpReturnCode              : RC_PPE_STATE_DATA_XIR
 90.66822|  FFDC                       : REG_SPRG0
 90.66823|     0000013C
 90.66826|------------------------------------------------
 90.66829|  HwpReturnCode              : RC_PPE_BASE_ADDR_XIXCR
 90.66830|  FFDC                       : PPE_BASE_ADDR
 90.66831|     00000000 00066010
 90.66835|------------------------------------------------
 90.66838|  HwpReturnCode              : RC_PPE_STATE_DATA_XIR
 90.66839|  FFDC                       : REG_XSR
 90.66839|     00031000
 90.66843|------------------------------------------------
 90.66846|  HwpReturnCode              : RC_PPE_STATE_DATA_XIR
 90.66846|  FFDC                       : REG_IAR
 90.66847|     FFF2D82C
 90.66851|------------------------------------------------
 90.66854|  HwpReturnCode              : RC_PPE_STATE_DATA_XIR
[... 932 lines cut ...]
 94.33873|  FFDC                       : PROC_CHIP_TARGET
 94.33877|     70753A6B 303A6E30 3A73303A 70303000 00000000 00000000 00000000 00000000
 94.33881|------------------------------------------------
 94.33884|  HwpReturnCode              : RC_SPCWKUP_EX_TIMEOUT
 94.33884|  FFDC                       : PPE_STATE_MODE
 94.33885|     00000004
 94.33889|------------------------------------------------
 94.33890|  Callout type             : Hardware Callout
 94.33892|  Target                   : Physical:/Sys0/Node0/Proc0/EQ3/EX1
 94.33893|  Deconfig State           : NO_DECONFIG
 94.33894|  GARD Error Type          : GARD_NULL
 94.33894|  Priority                 : SRCI_PRIORITY_HIGH
 94.33895|------------------------------------------------
 94.33896|  Callout type             : Procedure Callout
 94.33897|  Procedure                : EPUB_PRC_LVL_SUPP
 94.33898|  Priority                 : SRCI_PRIORITY_MED
 94.33898|------------------------------------------------
 94.33899|  Callout type             : Procedure Callout
 94.33900|  Procedure                : EPUB_PRC_HB_CODE
 94.33901|  Priority                 : SRCI_PRIORITY_MED
 94.33902|------------------------------------------------
 94.33902|  Hostboot Build ID: hostboot-0d0700f-pe7e80e1/hbicore.bin
 94.33903|================================================
106.08576|System shutting down with error status 0x613
```

dcrowell77, 2021-04-12:
I would suggest backing off the hcode level to the last known good one, then hostboot-binaries, then occ., then hostboot.

klauskiwi, 2021-04-12:
Yes, I was going to suggest a bisect of the offending commit.. and perhaps we could have CI enabled for the Blackbird system at some point?

dcrowell77, 2025-07-14:
That is quite a big range...  (202 commits?)  If you narrow it down more I can ask around internally to see if the offending commits ring any bells.  Given that the newer level works for us, even in SMT4 mode with opal I don't have any obvious explanations for why it doesn't work on this box.  Just for the heck of it, it might be worth trying the very latest level too in case there was just a bad window of time.

stewartsmith, 2025-07-14:
I've started bisecting between the two but still have a few steps before isolating it down.

stewartsmith, 2025-07-16:
```
:~/src/hcode$ git bisect log
git bisect start
# status: waiting for both good and bad commits
# good: [2bbe1ddd1aa3b5687ba9ed5aeddb221b7b9cdf9b] Release tag & head commit information updated for hw021820a.opmst
git bisect good 2bbe1ddd1aa3b5687ba9ed5aeddb221b7b9cdf9b
# status: waiting for bad commit, 1 good commit known
# bad: [5ed371eec5c7ed93b587da3b24ed10d7266972a8] Release tag & head commit information updated for hw030821a.opmst
git bisect bad 5ed371eec5c7ed93b587da3b24ed10d7266972a8
# good: [9f38a36dd5757beb11cdd1fae645837ce850f729] Release tag & head commit information updated for hw080420a.opmst
git bisect good 9f38a36dd5757beb11cdd1fae645837ce850f729
# good: [1e980f5884d51062a685efaa57e09b08c025e0d1] Merge "Revert "Updated devtoolset for RHEL7""
git bisect good 1e980f5884d51062a685efaa57e09b08c025e0d1
# bad: [e88e1c574da466c1cdb2675d7b1efdf412ecd90a] Release tag & head commit information updated for hw120920a.opmst
git bisect bad e88e1c574da466c1cdb2675d7b1efdf412ecd90a
# bad: [081fed299defa963aa7e21c8563d3a6a856a438e] Release tag & head commit information updated for hw110320a.opmst
git bisect bad 081fed299defa963aa7e21c8563d3a6a856a438e
# bad: [fad1a0f68bc3d532341e00680bac86e2d9dd7d19] Release tag & head commit information updated for hw102620a.opmst
git bisect bad fad1a0f68bc3d532341e00680bac86e2d9dd7d19
# bad: [a086ff7aa04958156559e93ab826f981b1c3c56b] PM: Process C405_ECC_UE FIR
git bisect bad a086ff7aa04958156559e93ab826f981b1c3c56b
# bad: [bc04b3a019b396db616d13607954f6152fb988f9] PFET: detect bad pfet through cme sgpe plumbling
git bisect bad bc04b3a019b396db616d13607954f6152fb988f9
# good: [071b0780cf71aeaf11fe2e4fa29efbe7d88e379e] Updated to devtoolset-8 for RHEL7
git bisect good 071b0780cf71aeaf11fe2e4fa29efbe7d88e379e
# first bad commit: [bc04b3a019b396db616d13607954f6152fb988f9] PFET: detect bad pfet through cme sgpe plumbling
```

Which narrowed it down to: https://github.com/open-power/hcode/commit/bc04b3a019b396db616d13607954f6152fb988f9
```
bc04b3a019b396db616d13607954f6152fb988f9 is the first bad commit
commit bc04b3a019b396db616d13607954f6152fb988f9
Author: Yue Du <daviddu@us.ibm.com>
Date:   Mon Sep 21 06:41:24 2020 -0500

    PFET: detect bad pfet through cme sgpe plumbling

    - checks for PFET Sense enable/disable ever being the same
    - check at STOP power on and power off
    - add FIT routine to peform check upon each interval
    - if checks fail, mark the core in a CME scratch reg and halt
    - CME halt is noticed by SGPE.  SGPE checks the CME scratch reg for
      either core to be marked.  Sets OCC LFIR[61] to id PFET issue.
    - Use CPMMR bit5:6 instead of scratch1 reg for pfet failure bits

    OCC LFIR bit will be used by p9_pm_callout to log PFET related
      errors by core. (not in this commit).

    Change-Id: I86a06ad951cbe879f06ace10cd40a0a484454f23
    CQ:SW508755
    Reviewed-on: http://rchgit01.rchland.ibm.com/gerrit1/104675
    Tested-by: Jenkins Server <pfd-jenkins+hostboot@us.ibm.com>
    Tested-by: PPE CI <ppe-ci+hostboot@us.ibm.com>
    Tested-by: Hostboot CI <hostboot-ci+hostboot@us.ibm.com>
    Reviewed-by: Gregory S Still <stillgs@us.ibm.com>
    Reviewed-by: RANGANATHPRASAD G. BRAHMASAMUDRA <prasadbgr@in.ibm.com>
    Reviewed-by: NAREN A DEVAIAH <naren.devaiah@in.ibm.com>
```

So it looks like `hw101920a.opmst` is the tagged version before that.

dcrowell77, 2025-07-22:
Note that the tags don't really mean much other than it was a daily grab.  All of the code would have passed tests before it was merged.

Based on pure speculation from that commit you posted, I wonder if your box could be detecting a PFET failure which would stop the PM engines and lead to a fail like the one you saw.  I would've thought that would get caught in step15 but only some of the complex is online there.  Also, if the behavior is to alert the OCC maybe that is what is preventing a startup.