open-power/op-build #4215: duplicate sysfs files created

shenki, 2021-07-19:
Multiple duplicate sysfs files are created at boot resulting in a lot of warnings.

> Linux skiroot 5.10.50-openpower1 #2 SMP Mon Jul 19 18:54:33 AEST 2021 ppc64le GNU/Linux
> OPAL skiboot-v6.8-26-g99c10192f 
> VERSION=2021.02.3-2-g2c7a998479

```
[    5.886569] at24 0-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    5.886928] sysfs: cannot create duplicate filename '/bus/nvmem/devices/module-vpd'
[    5.886932] CPU: 33 PID: 439 Comm: udevd Not tainted 5.10.50-openpower1 #2
[    5.886934] Call Trace:
[    5.886940] [c0000000101430a0] [c0000000002a0b90] dump_stack+0xac/0xe0 (unreliable)
[    5.886945] [c0000000101430e0] [c0000000001ff8f0] sysfs_warn_dup+0x70/0x88
[    5.886948] [c000000010143160] [c0000000001ffdc8] sysfs_do_create_link_sd.isra.2+0xd4/0x100
[    5.886953] [c0000000101431b0] [c0000000003380d8] bus_add_device+0x68/0xfc
[    5.886955] [c0000000101431f0] [c000000000335144] device_add+0x3f8/0x778
[    5.886958] [c0000000101432b0] [c0000000003d6e9c] nvmem_register+0x248/0x760
[    5.886961] [c0000000101433a0] [c0000000003d73fc] devm_nvmem_register+0x48/0x90
[    5.886964] [c0000000101433e0] [c0080000009e0894] at24_probe+0x5c4/0x678 [at24]
[    5.886968] [c000000010143630] [c0000000003af7e4] i2c_device_probe+0x15c/0x1e8
[    5.886970] [c000000010143670] [c000000000339470] really_probe+0x280/0x40c
[    5.886973] [c000000010143710] [c0000000003399fc] device_driver_attach+0x54/0x78
[    5.886976] [c000000010143750] [c000000000339b00] __driver_attach+0xe0/0xe4
[    5.886978] [c000000010143790] [c000000000336eec] bus_for_each_dev+0x90/0xd0
[    5.886981] [c0000000101437f0] [c000000000338b00] driver_attach+0x2c/0x40
[    5.886984] [c000000010143810] [c000000000338514] bus_add_driver+0x1bc/0x1fc
[    5.886986] [c0000000101438a0] [c00000000033a314] driver_register+0xe8/0x12c
[    5.886989] [c000000010143910] [c0000000003af4cc] i2c_register_driver+0x60/0x88
[    5.886992] [c000000010143940] [c0080000009e0b1c] at24_init+0x70/0x80 [at24]
[    5.886996] [c0000000101439a0] [c0000000000106d8] do_one_initcall+0x84/0x1c4
[    5.886999] [c000000010143a70] [c0000000000f8f7c] do_init_module+0x70/0x238
[    5.887002] [c000000010143b00] [c0000000000fac4c] load_module+0x19d8/0x2278
[    5.887004] [c000000010143cd0] [c0000000000fb708] __do_sys_finit_module+0xac/0xd0
[    5.887008] [c000000010143db0] [c00000000002147c] system_call_exception+0x140/0x164
[    5.887012] [c000000010143e20] [c00000000000c9c8] system_call_common+0xe8/0x218
[    5.887014] --- interrupt: c00 at 0x7fff898e99c4
[    5.887014]     LR = 0x7fff89a1a800
```

dcrowell77, 2021-07-19:
Seems like this might be better serviced over in the skiroot repo.

shenki, 2021-07-19:
It's a kernel issue, but someone in their infinite wisdom disabled the issue tracker for the kernel.

I guess it could be solved by disabling or renaming the nodes in skiboot, but I think then it's more of an op-build ecosystem issue if it crosses components.

dcrowell77, 2021-07-19:
@hegdevasant - FYI

hegdevasant, 2021-07-20:
> It's a kernel issue, but someone in their infinite wisdom disabled the issue tracker for the kernel.

I don't know who disabled issue tracker. Probably from day one it was disabled. Anyway I have enabled issue tracker.

> 
> I guess it could be solved by disabling or renaming the nodes in skiboot, but I think then it's more of an op-build ecosystem issue if it crosses components.

May be. I have to look into it.

-Vasant

hegdevasant, 2021-08-17:
This is tracked in upstream : https://bugzilla.kernel.org/show_bug.cgi?id=210749

-Vasant

hegdevasant, 2021-09-28:
I think its already fixed in upstream.
I could reproduce this issue on 5.10 kernel but I didn't hit this issue on 5.15-rc1.

-Vasant

closed 2021-09-28 by hegdevasant