open-power/op-build #4445: We do not have permission to access ibm-fw-proprietary

lili-lilili, 2021-11-30:
https://github.com/open-power/op-build/blob/abec1d3392a805366410a6aa637571a223ca87e5/openpower/package/ibm-fw-proprietary-p10/ibm-fw-proprietary-p10.mk#L8

when will this package open source to the public

dcrowell77, 2021-11-30:
This is by design.  That repository contains proprietary source code that is only available to certain users.  This logic is only needed to enable specific functions within the IBM firmware stack.  Which configuration are you attempting to build when you see a problem?

lili-lilili, 2021-12-06:
> 

We try to build master-p10 branch  with "p10ebmc_defconfig"

lili-lilili, 2021-12-06:
If i undefine this package, following error will reported during hostboot compilation:

 vpd_ecc_api_algorithms.C:30:10: fatal error: vpdecc.h: No such file or directory
 #include "vpdecc.h"  // vpd_create_ecc, vpdecc_check_data
          ^~~~~~~~~~
compilation terminated.
../../../src/build/mkrules/cc.rules.mk:99: recipe for target '../../../obj/modules/vpd/vpd_ecc_api_algorithms.o' failed
make[9]: *** [../../../obj/modules/vpd/vpd_ecc_api_algorithms.o] Error 1
../../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/CODE/BODY' failed
make[8]: *** [_BUILD/PASSES/CODE/BODY] Error 2
../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/SUBDIR/CODE/vpd' failed
make[7]: *** [_BUILD/SUBDIR/CODE/vpd] Error 2
make[7]: *** Waiting for unfinished jobs....

dcrowell77, 2021-12-06:
The solution is to create a custom configuration for your system that doesn't include the proprietary ECC algorithm.  Make this change to NOOP the ECC algorithm in the code and remove the dependency.

```
> # Enable the VPD ECC algorithms
> set COMPILE_VPD_ECC_ALGORITHMS
--
< # Disable the VPD ECC algorithms
< unset COMPILE_VPD_ECC_ALGORITHMS
```

If your company has a business relationship with IBM that includes sharing of this logic, please go through those channels to discuss a solution.

dcrowell77, 2022-03-25:
Closing per above comments.

closed 2022-03-25 by dcrowell77