open-power/op-build #5359: The console was added, but there is no output.

Emy-inspur, 2023-04-27:
On branch ibm-release-fw1030, commit 3c097c557727c3bb67d04313898f7890858de424
The following log is added to hostboot(commit:cc5f1cdd6ca59a8a83c9539c657f4837695722d3), but the host console does not output the log.
#include <console/consoleif.H>
CONSOLE::displayf(CONSOLE::DEFAULT, NULL, "Test log");

Attempts to add logs to the following hostboot files have no output:
/src/usr/pldm/extended/hb_fru.C
/src/usr/hdat/hdatiplparms.C

How to output console logs? Thankyou.

dcrowell77, 2023-04-27:
Is there any output on the console at all?  i.e. do you see the normal Hostboot startup messages and the isteps?  If not, I would guess you have something misconfigured on the BMC side.

How are you displaying the console?  We ssh to port 2200 on the BMC to see it.  The BMC also saves the console into /var/log/obmc-console.log .  I know there are configuration files on the BMC itself to control this but I don't have a lot of familiarity with the details.

Emy-inspur, 2023-04-28:
Checking the file of /var/log/obmc-console.log, I can see the normal Hostboot startup messages and the isteps, but I cannot see the log I added.
for example,  I add the log to cacheRemoteFruVpd() in hb_fru.C
errlHndl_t cacheRemoteFruVpd()
{
    using namespace TARGETING;

    PLDM_ENTER("cacheRemoteFruVpd");
   CONSOLE::displayf(CONSOLE::DEFAULT, NULL, "Test log"); //added

    errlHndl_t errl = nullptr;
   ....
}