open-power/op-build #5686: Blackbird fails to build pnor with BOOTKERNEL image too large

runlevel5, 2023-09-17:
I got following error when trying to build the firmware with GCC 11.4.0:

```
ERROR: PnorUtils::checkSpaceConstraints: Image provided (/home/tle/Work/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//zImage.epapr) has size (39321600) which is greater than allocated space (19660800) for section=BOOTKERNEL.  Aborting! at /home/tle/Work/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/PnorUtils.pm line 462.
```

The size is double the allocated space. I was thinking of increasing the physical size of BOOTKERNEL in the `output/build/openpower-pnor-3e9fa97dd8ac5e92de1e71178d41529e6ef9d9f1/p9Layouts/defaultPnorLayout_64.xml` but feeling something is not quite right.

So it's seems to me the XZ compression might have not worked as expected or some extra modules that might have occupied the space. The latter I could confirm not the case:

```
$ ls -lah output/target/lib64/modules/
total 0
drwxr-xr-x. 1 tle tle  38 Sep 17 12:56 .
drwxr-xr-x. 1 tle tle 634 Aug 29 20:13 ..
drwxr-xr-x. 1 tle tle 442 Sep 17 15:47 5.10.194-openpower1
```

The first hypothesis might be plausible. I've double checked the blackbird config and confirm `BR2_LINUX_KERNEL_XZ=y`.

Any suggestion how to troubleshoot this issue? Many thanks

dcrowell77, 2023-09-19:
Yes, you found the place, specifically it is the "ibs=19660800" option.  This is all part of the scripts under openpower-pnor.

rarbab, 2025-07-08:
`blackbird_defconfig` builds now. I think the root cause here was that buildroot started using `-O2` by default instead of `-Os`. See https://github.com/open-power/op-build/pull/6425#issuecomment-2956828740.