open-power/op-build #5845: petitboot：When the language is Chinese, no Chinese is displayed in petitboot

nancy-sunny, 2024-01-10:
Petitboot：Language
When the language is Chinese, no Chinese is displayed in petitboot.
Excuse me, how to configure to display Chinese？
I found that there is no locale folder under /usr/share/, how do I generate a locale folder？
thank you.

dcrowell77, 2024-01-10:
Please ask about this over in https://github.com/open-power/skiboot, that is where petitboot comes from

rarbab, 2024-01-11:
This isn't related to [skiboot](https://github.com/open-power/skiboot) at all, or even [petitboot](https://github.com/open-power/petitboot) itself per se, but about the internationalization of the overall image, so op-build is the right place to ask.

> Petitboot：Language When the language is Chinese, no Chinese is displayed in petitboot. Excuse me, how to configure to display Chinese？ I found that there is no locale folder under /usr/share/, how do I generate a locale folder？ thank you.

To populate /usr/share/locale with all the languages that petitboot lists in the "Language" menu, do you have something like this in your `.config`?

```
BR2_ENABLE_LOCALE_WHITELIST="C de en es fr it ja ko pt_BR ru zh_CN zh_TW"
BR2_GENERATE_LOCALE="en_US.UTF-8 de_DE.UTF-8 es_ES.UTF-8 fr_FR.UTF-8 it_IT.UTF-8 ja_JP.UTF-8 ko_KR.UTF-8 pt_BR.UTF-8 ru_RU.UTF-8 zh_CN.UTF-8 zh_TW.UTF-8"
BR2_SYSTEM_ENABLE_NLS=y
```

dcrowell77, 2024-01-11:
> the internationalization of the overall image

I suppose it depends on the scope of what you call "the image".  In my world the most important output is the host firmware files (pnor image, etc) and the Linux stuff is only consumed by the skiboot/petitboot portion of op-build.  I forget (if I ever knew) which layer settings like this are controlled. :-) 

Thanks for jumping in @rarbab

nancy-sunny, 2024-01-15:
Now it's ready to display. Thank you

closed 2024-01-16 by rarbab