open-power/op-build #5882: Sbe reference BR2_OPENPOWER_SIGNED_SECURITY_VERSION fail

lili-lilili, 2024-02-01:
https://github.com/open-power/op-build/blob/10fa0ab373a51ca9a7f6fcf3f2c0e79211804c51/openpower/configs/p10ebmc_defconfig#L90

Sbe want to reference BR2_OPENPOWER_SIGNED_SECURITY_VERSION, but it alway get a null value.

Sbe code:
https://github.com/open-power/sbe/blob/abc30fb8fe75dbd977496845c54c04f2c565f875/src/tools/scripts/signSbeImage#L241

Any suggestions?

mabaiocchi, 2024-02-07:
@lili-lilili , please excuse my delayed response.

The difference is that BR2_OPENPOWER_SIGNED_SECURITY_VERSION as defined in p10ebmc_defconfig is a buildroot variable and is not the same as the environment variable that signSbeImage script is expecting to use.

I am working with the SBE team on this issue and hope to have a fix available by the end of the week.

lili-lilili, 2024-02-18:
> The SBE team is verifying my changes. After they have done so, I will push up my changes as a pull request.

Now, i use this patch as a workaround:
```
diff --git a/openpower/package/sbe-p10/sbe-p10.mk b/openpower/package/sbe-p10/sbe-p10.mk
index e89bae33..edab67d5 100644
--- a/openpower/package/sbe-p10/sbe-p10.mk
+++ b/openpower/package/sbe-p10/sbe-p10.mk
@@ -28,6 +28,8 @@ define SBE_P10_BUILD_CMDS
                all
 endef
 
+export SIGNING_SECURITY_VERSION := $(BR2_OPENPOWER_SIGNED_SECURITY_VERSION)
+
 define SBE_P10_INSTALL_IMAGES_CMDS
        $(INSTALL) -D $(@D)/images/ipl_image_tool $(HOST_DIR)/usr/bin/
        python2 $(@D)/src/build/sbeOpDistribute.py  --sbe_binary_dir=$(STAGING_DIR)/sbe_binaries --img_dir=$(@D)/images --sbe_binary_filename $(BINARY_SBE_FILENAME)

```

mabaiocchi, 2024-02-20:
Pull Request with the fix: https://github.com/open-power/op-build/pull/5910

PR #5910 (2024-02-21): Add BR2_OPENPOWER_SIGNED_SECURITY_VERSION into sbe makefiles
This commit converts the buildroot variable BR2_OPENPOWER_SIGNED_SECURITY_VERSION into a local environment variable in specific spots of the sbe-p10 and sbe-odyssey makefiles.  This ensures that those packages are built with the same value as the rest of the build.

Resolves #5882
closed 2024-02-21 by lili-lilili